RetroPie/RetroPie-Setup #684: Duke Nukem 3D Mouse problems

PowerWiesel, 2015-02-23:
I can’t get my mouse running for aiming and shooting etc. For example it’s working wiith Quake out of the box.

In the menu (megaton edition) you can move the curser, but you can’t click anything.

Ingame there is also a moving courser, but you can’t shoot or aim.

I tried in_mouse 1 and MouseAiming = 1 and several little things that dind’t work

joolswills, 2015-02-23:
Thanks - Yeh I can confirm this.

agenttank, 2015-03-19:
same here, but i don't really care ;)
fresh install of retropie 2.6 full image

luxiouronimo, 2015-05-20:
i can confirm and add:

raspberry pi 2, installed retropie-v3.0beta2-rpi2.img, and install from yesterday's (5/19/15) binaries..

in the in-game menu the mouse works and i can click menu items, but in-game the mouse does not affect the aim/view or movement, even when toggling mouse move with the 'u' key.

clicking mouse1 does however shoot.

joolswills, 2015-05-21:
It's a known issue as above.

luxiouronimo, 2015-05-21:
to be clear, relating to v3.beta2-rpi2, you can click the in-game menu and that works, and in-game you can shoot.

joolswills, 2015-05-21:
yep - this is the issue. the mouse works in the menus, but not for aiming.

davidj1987, 2015-09-30:
Is Duke 3D eduke or a retroarch port? Is this issue related to this issue: https://github.com/libretro/Genesis-Plus-GX/issues/31

In this emulator mouse buttons work but the cursor doesn't move. :/

joolswills, 2015-09-30:
unrelated issue.

closed 2016-02-17 by joolswills

PowerWiesel, 2016-02-17:
Thank you very much:)

davidj1987, 2016-03-13:
I'm having mouse issues with eduke...is there something I need to enable?

HerbFargus, 2016-03-13:
This issue is closed. Please see the wiki (https://github.com/retropie/retropie-setup/wiki/Duke-Nukem-3D)  and use the forum for support

commit 335ed5fc (2016-02-17): switch eduke32 to sdl2 for jessie+ and enable opengl for X11 targets. Allow mali target. fixes #684