RetroPie/RetroPie-Setup #687: setting scaling_governor on Raspberry Pi 2 (runcommand.sh)

alberthorta, 2015-02-25:
Before running the emulator runcommand.sh is setting the cpu scaling_governor to "performance" to get the maximum performance of the CPU core, but the new Raspberry Pi 2 have 4 cores, and we are just setting the "performace" on the core 0.

I don't know if this can actually impact on the performance of the emulators, but seems reasonable.

I just copied and replaced the line : 

echo "performance" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null

into :

echo "performance" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null
echo "performance" | sudo tee /sys/devices/system/cpu/cpu1/cpufreq/scaling_governor >/dev/null
echo "performance" | sudo tee /sys/devices/system/cpu/cpu2/cpufreq/scaling_governor >/dev/null
echo "performance" | sudo tee /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor >/dev/null

And the line :

echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null

into : 

echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null
echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu1/cpufreq/scaling_governor >/dev/null
echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu2/cpufreq/scaling_governor >/dev/null
echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu3/cpufreq/scaling_governor >/dev/null

But, remember, this just make sense on Raspberry 2.

I still need to make some tests in order to see if this is helping at all with n64 emulator, that's the only emulator where I can feel that the speed is still not enough, even with Raspberry PI 2

--
Albert

joolswills, 2015-02-25:
thanks. certainly we should change all cores, but I wonder if we should do this by default anyway - it could cause the temperature to be higher, and certainly on the pi2, stuff seems to run fine in ondemand mode.

Will consider disabling this by default, but adding it as an option (and correcting it for the pi2).

thanks.

alberthorta, 2015-02-25:
A slight better solution (more elegant and cros compatible with RPI1):

Change
echo "performance" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null

To
for cpucore in /sys/devices/system/cpu/cpu?; do [[ -f "$cpucore" ]] || echo "performance" | sudo tee /sys/devices/system/cpu/cpu${cpucore:(-1)}/cpufreq/scaling_governor >/dev/null; done

And 

Change
echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor >/dev/null

To
for cpucore in /sys/devices/system/cpu/cpu?; do [[ -f "$cpucore" ]] || echo "ondemand" | sudo tee /sys/devices/system/cpu/cpu${cpucore:(-1)}/cpufreq/scaling_governor >/dev/null; done

joolswills, 2015-02-25:
thanks - yeh will be doing a loop to manage it (and likely will stick it in a function). Just need to sort out how to handle the settings.

joolswills, 2015-02-25:
just fyi, you could simplify it further by doing

```
for gov in /sys/devices/system/cpu/cpu[0-9]*/cpufreq/scaling_governor; do echo "performance" | sudo tee $gov >/dev/null; done
```

alberthorta, 2015-02-25:
Great! I prefer yours :-)

joolswills, 2015-02-25:
the * just in case there are more than 10 cores.. pi3 ? ;-) heh

joolswills, 2015-03-14:
this is now included. thanks.

closed 2015-03-14 by joolswills