RetroPie/RetroPie-Setup #688: Genesis Plus GX as default Emulator.

BlockABoots, 2015-02-25:
Is it possible to make Genplus-GX the default emulator for Megadrive, Sega CD and Masyer system games and just use PicoDrive for 32x games. Current DGen and Picodrive have alot of audio issues for Megadrive games either not sounding correct or missing audio altogether. Just load up Afterburner 2 for a good example,  there is massive audio issues with picodrive and DGens yet with Genplus-G its perfect

amigagamer, 2015-02-27:
GenplusGX libretro is by far the best mastersystem emulator ive found for RPI1 and 2. So id like to see this change for sega 8 bit also.  sound always seems in sync.and has the added benefit of the YM2413 synthesis in the core options, giving a significant music upgrade to games like Phantasy Star.

However as a megadrive emulator its a lot more resource hungry than picodrive or dgen so wouldnt be suitable for RPI1

gizmo98, 2015-03-01:
You can use Genesis-Plus-GX for megadrive/genesis emulation now:
https://github.com/petrockblog/RetroPie-Setup/commit/25296f50a70848f0d6e290340a006b51b53502e2

joolswills, 2015-03-21:
this emulator is now the default for the rpi2.

closed 2015-03-21 by joolswills