RetroPie/RetroPie-Setup #690: module/emulator name mismatches

joolswills, 2015-02-26:
Hi @petrockblog and @gizmo98 

I'd like to do some cleaning up of emulator names vs module names (and in some cases rom folders).

I think it would make it more obvious what emulator is being used (so users can search for info), and will avoid confusion. Seems this affects the libretro cores most.

an example is

genesislibretro

which has the description "GameGear LibretroCore"

but is actually Genesis-Plus-GX

another issue with the naming of libretro cores is their are very mixed:

libretrosomething
somethinglibretro
something

perhaps they should all include libretro as a suffix or prefix or not include it at all ? maybe a suffix of "-lr" would be enough to keep it short ? Seems most that have libretro in the name have it as a suffix currently sometimes with a hyphen sometimes without.

in terms of rom folder names, until we have https://github.com/Aloshi/EmulationStation/pull/391 I guess we go as we do currently, but renaming rom folders due to names of emulators changing such as "mastersystem-genesis" becoming perhaps "mastersystem-genesisplusgx" etc

I realise renaming stuff is a bit messy for users, but in the end we get a cleaner layout/system. I'm looking forward to when we can have a single rom folder per platform and handle which emulator is launched via emulationstation (as with the linked pull request).

Please add any thoughts below :)

gizmo98, 2015-02-26:
Sounds good. We could rename our modules so there is a clear structure. If we rename modules it is not clearly visible which libretro repo is used. But as you said. Names are already mixed and confusing so i don't see any problem doing this.

joolswills, 2015-02-27:
I see you made a pull request which makes some changes to a few different things including the renaming - I was hoping to have some feedback from @petrockblog  before going ahead with anything though. I'm also not sure we should be prefixing the other emulator names and I think @petrockblog needs to be in agreement on the naming before we do anything on this.

Would you be ok with splitting up your pull request, leaving the renaming for now, and just implementing the other emulator changes ?

The configuration menu work also actually conflicts with some code I have locally which does virtually the same thing (some different options and it's called the tools menu) - I guess we need more communication on this to we don't reproduce work - I probably should have mentioned I was working on this, although there is a ticket open regarding this. I don't have a problem merging yours, and I think we should use a folder parent to the roms folder and I have a few other things in mind, but they could be done later.

joolswills, 2015-02-27:
having a look at your changes, I'm not sure it solves the main issue I am raising - I still think the module name/filename should match - it's not just the filenames, but the internal module name that doesn't match the libretro emulator. so for example, mamelibretro is actually imame4all, so I think it should be called this.

I prefer the current naming over the new names I am afraid (sorry). But let's go over this further - and propose any changes first so we can all agree.

gizmo98, 2015-02-27:
Yeah. We can split out snes9x-next and genesis-plus-gx. The other things can be done later. Don't worry. I just wanted to start midnight commander from es and added the rest because i dislike esconfig module and all python scripts. Renaming and config menu took only 15 minutes. So no wast of time if we do something else. ;-)

joolswills, 2015-02-27:
I'm in full agreement regarding the config menu just a few differences in regards to what's in it and naming - I'm also keen for the removal of the python scripts. maybe it should be called "RetroPie" as a name - we could brand it then and have it the first menu that comes up perhaps ? 

So we can either commit your menu, and then adjust it, or I could commit mine, but incorporate your stuff into it also (like mc etc).

so what do people think for a name

Configuration
RetroPie
Tools

Something else ?
?

gizmo98, 2015-02-27:
Renaming module names would be the nextstep. But i no interest today opening all modules and rename all functions and module names ;-)

joolswills, 2015-02-27:
np :)

I'm sort of thinking now we should rename as few as possible, to not break the existing structure too much. so maybe it's just enough to just add a -lr or something to the name if they conflict with non retroarch emulator names? and make sure that the names better match the retroarch emulators (mame -> imame4all etc).

joolswills, 2015-02-27:
(or if we are to prefix/suffix - just do the libretro cores, and leave everything else as is). Anyway, enough of me dumping my brain on github for now.

gizmo98, 2015-02-28:
I thought about it. A own section named retropie with the mushroom berry logo should be nice (http://blog.petrockblock.com/forums/topic/mushberry-retro/). bashwelcometweak already uses mushroom berry. I don't know where it originated. Copyright?

petrockblog, 2015-02-28:
I have thought about a generic naming scheme for the module IDs as well as for the rom folders. Here are my suggestions:

### Module Naming Conventions for LibretroCores

To make the module names unique it should be obvious when a Libretro core is used. To keep names as short as possible I suggest to use the abbreviation "`lr`" with the following naming scheme for Libretro cores:

`<rp_module_id> := lr-<original core name>`

### ROM Folder Naming Convention:

rom subfolder names `<rom folder name>`:
`<rom folder name> := <system>-<rp_module_id>`

`<rp_module_id> `: emulators like "atari800", "pifba" etc. or Libretro core (see thoughts about Libretro core naming scheme above) 
`<system>`: fba, neogeo, snes, gbc, ...

Overall, this scheme allows the users to clearly see the emulated system together with the used emulator and it is also obvious which module is responsible the folder.

What do you think?

joolswills, 2015-02-28:
that works for me, and is pretty much how I was thinking - I like the abbreviated lr - we don't want overly verbose module ids.

and at some point we will hopefully be able to rid ourselves of multiple rom directories completely by having the emulator configurable from emulationstation.

petrockblog, 2015-02-28:
@gizmo98: I once had the idea to use the Mushberry logo as "official" RetroPie logo, but I was very quickly notified by the Raspberry Pi Foundation that the Mushberry logo violates the Raspberry Pi trademark. However, I am currently planing something in that direction so I just would not use any logo for now in such a menu. 
Generally, I agree that the current "config" menu is not a well ordered place and I think it is worth spending some thoughts on that ...

gizmo98, 2015-02-28:
This is really sad because it is such a friendly and non-technical logo. Is it because the top is raspberry pi foundations logo? Would it be sufficient to use another raspberry drawing? We could start a contest and use the best looking logo.

joolswills, 2015-02-28:
I added an initial commit for the configuration menu

https://github.com/petrockblog/RetroPie-Setup/commit/34776f10f5089124c4e6bef1b484aeb985ef50a9

based on some code i had + some bits from @gizmo98 

now we just need it to be pretty - and I guess some descriptions would be good too.

joolswills, 2015-03-03:
I'm working on the renaming - just a heads up so we don't duplicate work. When I have it done, i'll do a pull request and please review it to spot any mistakes etc :)

petrockblog, 2015-03-03:
Thanks for the heads-up  - all right then!

gizmo98, 2015-03-03:
Thanks jools!

joolswills, 2015-03-04:
I've done the initial rename (https://github.com/joolswills/RetroPie-Setup/commit/2bde7a261cbd94b62ed02ff712163be7fb1c2125), but I'm not happy about the rom folder naming - the current thing with multiple rom folders is just rather messy, and confusing. I'd like to sort it out, and I have an idea to do it, with some additional configs, which would allow users to choose a default emulator, as well as switch which emulator is used per rom - so we just have a single rom folder per platform (apart from the case where users may need different romsets like with mame etc). 

It's going to be some work I guess, but I think it needs doing to make sure we don't just have more and more rom folders for each new emulator that is requested.

I'm also thinking to re-do some of the recent save-game stuff, as it could be de-duplicated somewhat. Currently savegames go to the rom folders, but perhaps we should have a separate location - thoughts ?

gizmo98, 2015-03-08:
Could you please remove the iniSet "savefile_directory" "*, " iniSet "savestate_directory" "*" as well? Actually i thought the "sram cannot be saved" message is a common issue and it was gone with these changes. But it seems to be a pocketsnes issue which persists. You could also remove all --appenconfig lines. snes9x-next and genesis-plus-gx for example already do not need appendconfig.

I would do it myself but i don't want to interfere your work.

joolswills, 2015-03-08:
sure - actually I planned to de-duplicate that stuff - are you saying none of the lines are needed at all ?

joolswills, 2015-03-08:
will sort the appendconfig also.

gizmo98, 2015-03-08:
Non of them are needed. There are one or two emulators (pocketsnes especially) with sram errors. But theses lines will not fix these issue. Pocketsnes wasn't updated for a while. Could be a compatibility issue with new retroarch builds.

joolswills, 2015-03-08:
and input_remapping_directory ?

gizmo98, 2015-03-08:
input_remapping is a really nice feature for different systems. The default button mapping of input autoconfig files does not always fit systems like megadrive, gameboy or nes. With remapping i can swap A/B buttons for example for only one system. So keep a remapping file in every system directory should be right.

joolswills, 2015-03-08:
k. thanks.

joolswills, 2015-03-08:
https://github.com/joolswills/RetroPie-Setup/commit/65c8f43b716f6ae2e41e70e15596b3dfc82f89e5 - looks ok ?

gizmo98, 2015-03-08:
Yes. Looks ok!

joolswills, 2015-03-08:
i committed that to master instead - as I decided to re-do my branch rename stuff (forgot to include some changes, so want to merge some commits together)

joolswills, 2015-03-08:
ok new branch is here https://github.com/joolswills/RetroPie-Setup/tree/lr_module_and_emulator_rework - the renaming is complete, and the old install directories are removed during the configure stage.

Have not done anything with rom folders yet as I am going to rework that completely as mentioned. I will hopefully have some time this coming week for that.

joolswills, 2015-03-08:
wip:

http://malus.exotica.org.uk/~buzz/def1.png
http://malus.exotica.org.uk/~buzz/def2.png

petrockblog, 2015-03-09:
That looks very promising! Looking at it I found all options that I would have expected.

gizmo98, 2015-03-09:
@joolswills 
You are awesome! That looks really good.

joolswills, 2015-03-09:
getting close now in terms of functionality. There are likely to be plenty of bugs. I have posted on the forum regarding some user testing and will provide some test images for those that want to have a go and feed back any issues.

joolswills, 2015-03-13:
Closing as this is all merged. Cheers!

closed 2015-03-13 by joolswills