RetroPie/RetroPie-Setup #691: [Feature Request] : Use "make -j 4" when building packages from source on Pi 2.

andrewsil1, 2015-02-26:
Using all 4 cores during a "make" has a huge (positive) impact on overall build completion time.  Is this something that perhaps should be generally used in the build scripts when we know we're running on a Pi 2?

joolswills, 2015-02-26:
https://github.com/petrockblog/RetroPie-Setup/issues/663

I'd rather not enable anything like that by default yet, as there have been some emulators that break during parallel compilation (some infact only failed occasionally which made it more annoying). I have been going through fixing up makefiles etc, but just to avoid breaking things I still think it's better to enable it manually.

andrewsil1, 2015-02-26:
Fair enough. At least it's been considered. :-)

closed 2015-02-26 by andrewsil1

joolswills, 2015-02-26:
feel free to use the MAKEFLAGS parameter though and let us know if you notice any build failures. If you want to build even quicker you can set up distcc

When doing a full build so as to update the binaries, I build in parallel but also with distcc in an emulated chroot environment. Worth trying if you really want stuff built fast :)

commandline for that on my system looks something like

```
sudo PATH="/usr/lib/distcc:$PATH" MAKEFLAGS="-j8 PATH=/usr/lib/distcc:$PATH" __platform=rpi2 ./retropie_setup.sh
```

platform variable is only needed if it can't be autodetected or in my chroot case, where I build for both rpi1 and rpi2. I have distcc on the host machine, so the emulated pi is using 8 threads for preprocessing, and the host is compiling with 8.