RetroPie/RetroPie-Setup #694: Possible Daphne?

space-invader, 2015-02-27:
Hi.
Could it be posible to run Daphne (Laser Disc emulator) to play Dragon’s Lair and Space Ace?

There is a Linux versión, but I don’t know whether it could be run on RPi:

http://www.daphne-emu.com/download/daphne-1.0beta-linux.tar.gz

joolswills, 2015-02-28:
the one you link can't be - it is a binary build for x86.

Source is here http://www.daphne-emu.com/download/daphne-1.0-src.tar.bz2 but ive not looked at it.

HerbFargus, 2015-08-02:
This is the latest on the daphne project: https://www.raspberrypi.org/forums/viewtopic.php?t=64826&p=512530 as it currently stands it doesn't look like the licenses would be compatible with the RetroPie project- the developer currently wishes to retain rights without releasing the source code though he may be open to releasing it the future.

HerbFargus, 2016-02-15:
Incomplete Module, just for reference. See also forum post: http://blog.petrockblock.com/forums/topic/dragons-lair-space-ace-etc-anyone-interested-in-daphne-laserdisc-system/

still needs some makefile fixes to build the main binary, and of course lots of configuration fixes

```
#!/usr/bin/env bash

# This file is part of The RetroPie Project
# 
# The RetroPie Project is the legal property of its developers, whose names are
# too numerous to list here. Please refer to the COPYRIGHT.md file distributed with this source.
# 
# See the LICENSE.md file at the top-level directory of this distribution and 
# at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
#

rp_module_id="daphne"
rp_module_desc="Daphne - Laserdisc Emulator"
rp_module_menus="4+"

function depends_daphne() {
    getDepends libvorbis-dev libglew-dev
}

function sources_daphne() {
    gitPullOrClone "$md_build" https://github.com/Shmoopty/daphne-pi.git
}

function build_daphne() {
    # Build vldp2 lib  
    pushd ./src/vldp2/
    ./configure --prefix="$md_inst"
    make clean
    make -f Makefile.vars.linux_rpi2
    popd
    pushd ./src/
    rm -f ./Makefile.vars
    ln -s ./Makefile.vars.linux_rpi2 ./Makefile.vars
    make clean
    make
    popd
}

function install_daphne() {
    md_ret_files=(
        'daphne'
        'daphne.bin'
        '/supplementary/dapinput.ini'
        '/supplementary/README.txt'
        'LICENSE'
    )
}

function configure_daphne() {
    mkRomDir "daphne"
    ensureSystemretroconfig "daphne"

    addSystem 1 "$md_id" "daphne" "$md_inst/daphne.bin lair noldp -nohwaccel"
}
```

commit cf3d21df (2016-02-17): added daphne emulator to experimental - thanks to shmoopty - fixes #694
closed 2016-02-17 by joolswills