RetroPie/RetroPie-Setup #695: Add snes9x-next retroarch core

cacophony555, 2015-02-27:
Please add the snes9x-next retroarch core to RetroPie:
https://github.com/libretro/snes9x-next

It has far more accurate sound emluation than the other existing options and the RPi 2 is powerful enough to run it well.

gizmo98, 2015-03-01:
You can test it now:
https://github.com/petrockblog/RetroPie-Setup/commit/80a12bfd0f2a78f3302e700b42cd1dfe111381e0

jwkassem, 2015-03-02:
I cannot see the option for updating on rpi2 the script makes reference rpi1 only. is this correct? The option also does not appear from setup. this also running update script and rebooting. The menu list 221 is not available on my rpi2.

joolswills, 2015-03-02:
did you look under the experimental menu ?

jwkassem, 2015-03-02:
Thanks Running now.  It's console has a few warnings. I'm running from the stock 2.6 rpi2 image. Do I need to update anything for the build environment?

joolswills, 2015-03-02:
not sure what you mean ?

jwkassem, 2015-03-02:
i know when i build on my mac. it will say install say some extras on order to compile programs. 

sorry forgive me if i'm out of line here. I noticed some chatter on rpi forums about reicast emu for the rpi2. have had a look at this? Also i noticed some talk on cps3 roms for fba.

joolswills, 2015-03-02:
the retropie modules/scripts for each emulator should handle any needed dependencies.

for emulator requests please open a new ticket (or discuss first on the forum). Thanks.

joolswills, 2015-03-14:
snes9x-next is now included by default (for rpi2).

closed 2015-03-14 by joolswills