RetroPie/RetroPie-Setup #696: Yabause [Sega Saturn Emulator], Possible?

InsecureSpike, 2015-02-27:
ok, so now we have the luxury of more ram with the rPi2 are we able to look into this?
i would try myself but i have no idea how to go about any of the installation process?

heres a link I've found...

http://www.0001.ca/raspberrypi/oicrtpi/usr/portage/games-emulation/yabause/

frthery, 2015-02-27:
Hi, i have already tested yabause core on my pi, it's very, very, very slow!
I am curious to see how it's running on pi2, but i am sure we can't expect a correct speed :/

module: https://github.com/frthery/ES_RetroPie/blob/master/RetroPie-Setup/scriptmodules/libretrocores/a_yabauselibretro.sh

gizmo98, 2015-02-27:
Yabause runs with 1fps on a pi1. I don't expect it running much faster on a pi2.

InsecureSpike, 2015-02-27:
oh, thank you for your help fellas!,

closed 2015-03-26 by joolswills