RetroPie/RetroPie-Setup #697: Installation of XBox 360 controller fails

smoky19x, 2015-02-27:
Hello,

i noticed that on installing the XBox 360 controller via retropie_setup.sh, the modification of /etc/rc.local fails because the check in line 8 of /scriptmodules/supplementary/xboxdrv.sh ("if grep -q "xboxdrv" /etc/rc.local; then") does not work correctly. The check in previous versions ("if [[ -z $(cat /etc/rc.local | grep "xboxdrv") ]]; then") worked.

Could you please fix the check?

I would also suggest, instead of modifying the /etc/rc.local file, maybe it is better to install the init.d script as described here: https://github.com/petrockblog/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller

Regards,
Daniel

joolswills, 2015-03-03:
Have fixed the bug thanks. I agree that the init.d script looks like a better method - I don't have a 360 controller for testing though, so if you are happy to test my changes, I'll implement it. Cheers.

joolswills, 2015-03-03:
have you tested the init.d method vs the /etc/rc.local btw ?

smoky19x, 2015-03-03:
Yes, i use the init.d script for setting up my XBox 360 controllers. I prefer it over the rc.local method, because it runs as a daemon and is configurable by editing the file /etc/default/xboxdrv. The only thing i noticed, is that xboxdrv itself sometimes creates something like /dev/input/js2 and js3 instead of /dev/input/js0 and js1. But if i stop and start the sript again, it is alright. If you want to, i can investigate the reason for this a bit more.

closed 2015-03-03 by smoky19x

joolswills, 2015-03-03:
the script has some code to rename if they already exist - but it could be better just to remove them first perhaps.