RetroPie/RetroPie-Setup #698: Building of basilisk fails

smoky19x, 2015-02-27:
Hello,

if i try to install the Macintish emulator basilsik from source on a rpi2, it fails at the moment with the following error from the log:

"
...
sigsegv.cpp:2610:30: error: ‘SIGSEGV_FAULT_HANDLER_ARGLIST’ was not declared in this scope
sigsegv.cpp:2611:1: error: expected ‘,’ or ‘;’ before ‘{’ token
Makefile:189: recipe for target 'obj/sigsegv.o' failed
make: *** [obj/sigsegv.o] Error 1
/home/pi/RetroPie-Setup
Could not successfully build_basilisk Macintosh emulator (/home/pi/RetroPie-Setup/tmp/build/basilisk/BasiliskII/src/Unix/BasiliskII not found).
"

Maybe this can be fixed.

Regards,
Daniel

smoky19x, 2015-02-27:
And here is the full log:

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for Macintosh emulator
= = = = = = = = = = = = = = = = = = = = =

/home/pi/RetroPie-Setup/tmp/build/basilisk /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for Macintosh emulator
= = = = = = = = = = = = = = = = = = = = =

macemu-master/.gitignore
macemu-master/BasiliskII/
macemu-master/BasiliskII/BasiliskII.spec
macemu-master/BasiliskII/COPYING
macemu-master/BasiliskII/ChangeLog
macemu-master/BasiliskII/INSTALL
macemu-master/BasiliskII/Makefile
macemu-master/BasiliskII/README
macemu-master/BasiliskII/TECH
macemu-master/BasiliskII/TODO
macemu-master/BasiliskII/src/
macemu-master/BasiliskII/src/AmigaOS/
macemu-master/BasiliskII/src/AmigaOS/BasiliskII.info
macemu-master/BasiliskII/src/AmigaOS/Makefile
macemu-master/BasiliskII/src/AmigaOS/asm_support.asm
macemu-master/BasiliskII/src/AmigaOS/audio_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/clip_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/ether_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/extfs_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/main_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/prefs_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/prefs_editor_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/scsi_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/serial_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/sys_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/sysdeps.h
macemu-master/BasiliskII/src/AmigaOS/timer_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/user_strings_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/user_strings_amiga.h
macemu-master/BasiliskII/src/AmigaOS/video_amiga.cpp
macemu-master/BasiliskII/src/AmigaOS/xpram_amiga.cpp
macemu-master/BasiliskII/src/BeOS/
macemu-master/BasiliskII/src/BeOS/Makefile
macemu-master/BasiliskII/src/BeOS/SheepDriver/
macemu-master/BasiliskII/src/BeOS/SheepDriver/Makefile
macemu-master/BasiliskII/src/BeOS/SheepDriver/sheep_driver.c
macemu-master/BasiliskII/src/BeOS/SheepDriver/sheep_driver.h
macemu-master/BasiliskII/src/BeOS/SheepNet/
macemu-master/BasiliskII/src/BeOS/SheepNet/Makefile
macemu-master/BasiliskII/src/BeOS/SheepNet/sheep_net.cpp
macemu-master/BasiliskII/src/BeOS/SheepNet/sheep_net.h
macemu-master/BasiliskII/src/BeOS/about_window.cpp
macemu-master/BasiliskII/src/BeOS/about_window.h
macemu-master/BasiliskII/src/BeOS/audio_beos.cpp
macemu-master/BasiliskII/src/BeOS/clip_beos.cpp
macemu-master/BasiliskII/src/BeOS/ether_beos.cpp
macemu-master/BasiliskII/src/BeOS/extfs_beos.cpp
macemu-master/BasiliskII/src/BeOS/main_beos.cpp
macemu-master/BasiliskII/src/BeOS/prefs_beos.cpp
macemu-master/BasiliskII/src/BeOS/prefs_editor_beos.cpp
macemu-master/BasiliskII/src/BeOS/scsi_beos.cpp
macemu-master/BasiliskII/src/BeOS/serial_beos.cpp
macemu-master/BasiliskII/src/BeOS/sys_beos.cpp
macemu-master/BasiliskII/src/BeOS/sysdeps.h
macemu-master/BasiliskII/src/BeOS/timer_beos.cpp
macemu-master/BasiliskII/src/BeOS/user_strings_beos.cpp
macemu-master/BasiliskII/src/BeOS/user_strings_beos.h
macemu-master/BasiliskII/src/BeOS/video_beos.cpp
[... 1122 lines cut ...]
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c ./../slot_rom.cpp -o obj/slot_rom.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c ./../rsrc_patches.cpp -o obj/rsrc_patches.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c ./../emul_op.cpp -o obj/emul_op.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c ./../macos_util.cpp -o obj/macos_util.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c ./../xpram.cpp -o obj/xpram.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/include/SDL -D_GNU_SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c xpram_unix.cpp -o obj/xpram_unix.o
g++ -I./../include -I./. -I. -I./../uae_cpu -I./../slirp -DHAVE_CONFIG_H  -DOS_linux -DFPU_IEEE -D_REENTRANT -DDATADIR=\"/opt/retropie/emulators/basilisk/share/BasiliskII\" -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -I/usr/ [... cut ...] SOURCE=1 -D_REENTRANT -fno-merge-constants -fno-gcse-sm  -c sigsegv.cpp -o obj/sigsegv.o
sigsegv.cpp:2610:30: error: ‘SIGSEGV_FAULT_HANDLER_ARGLIST’ was not declared in this scope
sigsegv.cpp:2611:1: error: expected ‘,’ or ‘;’ before ‘{’ token
Makefile:189: recipe for target 'obj/sigsegv.o' failed
make: *** [obj/sigsegv.o] Error 1
/home/pi/RetroPie-Setup
Could not successfully build_basilisk Macintosh emulator (/home/pi/RetroPie-Setup/tmp/build/basilisk/BasiliskII/src/Unix/BasiliskII not found).

joolswills, 2015-02-28:
I am unable to reproduce here, but I get a different output from configure in regards to `checking whether we can skip instruction in SIGSEGV handler` - I assume as with the other ticket you are in a qemu emulated chroot environment - so it may be related to your set up there. Please test first on the real hardware, as I can't help with issues that might be specific to your chroot / and versions of software you have on your host etc.

smoky19x, 2015-02-28:
OK, thanks for the info. I will close the ticket.

Is there a small howto or similar, that describes howto setup a chroot and/or distcc environment for developing and testing? How is the process to create the sdcard image that is provided for download? When and how are the binaries updatet, that could be installed that can be installed by the RetroPie-Setup script?
I like this project very much and can imagine to help a bit, like testing, reporting and maybe fixing bugs...

closed 2015-02-28 by smoky19x

joolswills, 2015-02-28:
there is some code in the repository for generating of the image etc. Check in tools/sdcard

the binaries are updated when stuff changes that is significant or I get time to do it. Currently it's at least once a month.

there are existing howto's on using chroot/distcc with the pi, but there is none written specifically for this project. Once distcc is set up (http://www.openframeworks.cc/setup/raspberrypi/Raspberry-Pi-DISTCC-guide.html) you can build in a chroot with something like

sudo PATH="/usr/lib/distcc:$PATH" MAKEFLAGS="-j8 PATH=/usr/lib/distcc:$PATH" __platform=rpi2 ./retropie_setup.sh (or retropie_packages.sh)

also check the chroot.sh script in tools/sdcard for the script I use to mount/use my chroot

joolswills, 2015-02-28:
I do intend at some point to automatically build/update the binaries - but I have been very busy on other ongoing development currently.