RetroPie/RetroPie-Setup #699: Building of picodrive fails

smoky19x, 2015-02-27:
Hello,

if i try to install picodrive from source on a rpi2, it fails at the moment with the following error from the log:

"
...
In file included from platform/libretro/libretro.c:27:0:
./pico/pico_int.h:43:36: fatal error: ../cpu/cyclone/Cyclone.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'platform/libretro/libretro.o' failed
make: *** [platform/libretro/libretro.o] Error 1
/home/pi/RetroPie-Setup
Could not successfully build_picodrive Genesis LibretroCore Picodrive (/home/pi/RetroPie-Setup/tmp/build/picodrive/picodrive_libretro.so not found).
"

Maybe this can be fixed.

Regards,
Daniel

joolswills, 2015-02-27:
please can you give me the complete log from the point of getting the sources etc.

smoky19x, 2015-02-27:
OK, here it is:

/home/pi/RetroPie-Setup/tmp/build/picodrive /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for Genesis LibretroCore Picodrive
= = = = = = = = = = = = = = = = = = = = =

picodrive-master/.gitignore
picodrive-master/.gitmodules
picodrive-master/AUTHORS
picodrive-master/COPYING
picodrive-master/ChangeLog
picodrive-master/Makefile
picodrive-master/Makefile.libretro
picodrive-master/README
picodrive-master/configure
picodrive-master/cpu/
picodrive-master/cpu/DrZ80/
picodrive-master/cpu/DrZ80/DrZ80.txt
picodrive-master/cpu/DrZ80/drz80.h
picodrive-master/cpu/DrZ80/drz80.s
picodrive-master/cpu/cyclone/
picodrive-master/cpu/cyclone_config.h
picodrive-master/cpu/cz80/
picodrive-master/cpu/cz80/cz80.c
picodrive-master/cpu/cz80/cz80.h
picodrive-master/cpu/cz80/cz80_op.c
picodrive-master/cpu/cz80/cz80_opCB.c
picodrive-master/cpu/cz80/cz80_opED.c
picodrive-master/cpu/cz80/cz80_opXY.c
picodrive-master/cpu/cz80/cz80_opXYCB.c
picodrive-master/cpu/cz80/cz80jmp.c
picodrive-master/cpu/cz80/cz80macro.h
picodrive-master/cpu/debug.c
picodrive-master/cpu/debug.h
picodrive-master/cpu/debug_net.c
picodrive-master/cpu/debug_net.h
picodrive-master/cpu/drc/
picodrive-master/cpu/drc/cmn.c
picodrive-master/cpu/drc/cmn.h
picodrive-master/cpu/drc/emit_arm.c
picodrive-master/cpu/drc/emit_x86.c
picodrive-master/cpu/fame/
picodrive-master/cpu/fame/fame.h
picodrive-master/cpu/fame/famec.c
picodrive-master/cpu/fame/famec_opcodes.h
picodrive-master/cpu/musashi/
picodrive-master/cpu/musashi/Makefile
picodrive-master/cpu/musashi/m68k.h
picodrive-master/cpu/musashi/m68k_in.c
picodrive-master/cpu/musashi/m68kconf.h
picodrive-master/cpu/musashi/m68kcpu.c
picodrive-master/cpu/musashi/m68kcpu.h
picodrive-master/cpu/musashi/m68kdasm.c
picodrive-master/cpu/musashi/m68kmake.c
picodrive-master/cpu/musashi/readme.txt
picodrive-master/cpu/sh2/
picodrive-master/cpu/sh2/compiler.c
picodrive-master/cpu/sh2/compiler.h
picodrive-master/cpu/sh2/mame/
picodrive-master/cpu/sh2/mame/sh2.c
picodrive-master/cpu/sh2/mame/sh2dasm.c
picodrive-master/cpu/sh2/mame/sh2dasm.h
picodrive-master/cpu/sh2/mame/sh2pico.c
picodrive-master/cpu/sh2/sh2.c
picodrive-master/cpu/sh2/sh2.h
picodrive-master/jni/
picodrive-master/jni/Android.mk
picodrive-master/jni/Application.mk
[... 297 lines cut ...]
picodrive-master/zlib/trees.h
picodrive-master/zlib/uncompr.c
picodrive-master/zlib/zconf.h
picodrive-master/zlib/zlib.h
picodrive-master/zlib/zutil.c
picodrive-master/zlib/zutil.h
You need to run this command from the toplevel of the working tree.
You need to run this command from the toplevel of the working tree.
/home/pi/RetroPie-Setup
/home/pi/RetroPie-Setup/tmp/build/picodrive /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Building Genesis LibretroCore Picodrive
= = = = = = = = = = = = = = = = = = = = =

rm -f PicoDrive platform/common/mp3.o platform/common/mp3_dummy.o zlib/gzio.o zlib/inffast.o zlib/inflate.o zlib/inftrees.o zlib/trees.o zlib/deflate.o zlib/crc32.o zlib/adler32.o zlib/zutil.o zlib/compress.o zlib/uncompr.o unzip/unzip.o unzip/unzip_stream.o pico/pico.o pico/cart.o pico/memory.o pico/state.o pico/sek.o pico/z80if.o pico/videoport.o pico/draw2.o pico/draw.o pico/mode4.o pico/misc.o pico/eeprom.o pico/patch.o pico/debug.o pico/media.o pico/sms.o pico/cd/mcd.o pico/cd/cd_memory.o pico/cd/cd_sek.o pico/cd/cdc.o pico/cd/cdd.o pico/cd/cd_image.o pico/cd/cue.o pico/cd/gfx.o pico/cd/gfx_dma.o pico/cd/cd_misc.o pico/cd/pcm.o pico/32x/32x.o pico/32x/32x_memory.o pico/32x/32x_draw.o pico/32x/sh2soc.o pico/32x/pwm.o pico/pico/pico_pico.o pico/pico/pico_memory.o pico/pico/xpcm.o pico/carthw/carthw.o pico/carthw/svp/svp.o pico/carthw/svp/svp_memory.o pico/carthw/svp/ssp16.o pico/sound/sound.o pico/sound/sn76496.o pico/sound/ym2612.o pico/sound/mix.o cpu/fame/famec.o cpu/cz80/cz80.o cpu/drc/cmn.o cpu/sh2/sh2.o cpu/sh2/mame/sh2pico.o
rm -f -r .opk_data
cc -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -marm -DNO_ZLIB -fPIC -Wall -ggdb -falign-functions=2 -I. -DINLINE=inline -O2 -DNDEBUG -ffunction-sections -D_SVP_DRC -DEMU_C68K -D_USE_DRZ80 -DDRC_SH2   -c -o platform/libretro/libretro.o platform/libretro/libretro.c
In file included from platform/libretro/libretro.c:27:0:
./pico/pico_int.h:43:36: fatal error: ../cpu/cyclone/Cyclone.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'platform/libretro/libretro.o' failed
make: *** [platform/libretro/libretro.o] Error 1
/home/pi/RetroPie-Setup
Could not successfully build_picodrive Genesis LibretroCore Picodrive (/home/pi/RetroPie-Setup/tmp/build/picodrive/picodrive_libretro.so not found).

joolswills, 2015-02-27:
ok thanks. and you are building this directly on the pi2 and not in a chroot environment ?

If so, it looks like the code to detect a chroot environment isn't working and being triggered which it shouldn't (we have code to download data differently in an emulated chroot environment, which doesn't work with picodrive).

I may have to change the code to sort this - could you detail your set up ? are you using the image or installing on raspbian. I'd be interested in a copy of your image to see why it thinks you are in a chroot.

smoky19x, 2015-02-28:
Sorry, you are right. It was build in a chroot environment with an image of the sdcard version. I did not know that it makes such a difference.

Could you please explain in short why it is a difference for compiling if it is made in a chroot environment?

closed 2015-02-28 by smoky19x

joolswills, 2015-02-28:
aah - always good to mention details like this :) 

I do most of the development in an emulated chroot, so it handles this sirtuation, including setting the emulated arm cpu etc depending if building for rpi1 or rpi2.

unfortunately though, some stuff can lock up in this environment and git seems to do this often (and cmake https://bugs.launchpad.net/qemu/+bug/955379). so to avoid this, instead of using git, to download from github a tar is downloaded if in a chroot environment. this doesn't work with picodrive though  as it uses git submodules for the cyclone emulator.

to force the code to use git you can edit scriptmodules/helpers.sh and comment out the code in function gitPullOrClone that uses tar. At some point I need to do an exclusion for picodrive for this (the only emulator currently affected)