RetroPie/RetroPie-Setup #707: Pifba-libreto Insert a coin

expandables2, 2015-03-02:
Hi i am trying to play a game on pifba-libreto. The games comes on but i cannot insert a coin. I am using a PS3 controller

joolswills, 2015-03-02:
Please can you give some examples as I know I have been able to insert a coin on this emulator using the keyboard - I have read elsewhere there have been problems on certain games - is that the same for you, or you can't insert coins on any game ? Can you insert coins using the keyboard or just not with the controller (is it just that controller or have you tried others)?

Unfortunately I don't have any ps3 controller for testing that specific example.

expandables2, 2015-03-02:
I have input_enable_hotkey_btn = "0"
input_exit_emulator_btn = "3"

I tested on many games none work.

expandables2, 2015-03-02:
Sorry  it works now here is my config.

input_device = "Sony PLAYSTATION(R)3 Controller"
input_driver = "udev"
input_player1_joypad_index = "0"
input_player1_a_btn = "13"
input_player1_b_btn = "14"
input_player1_x_btn = "12"
input_player1_y_btn = "15"
input_player1_l_btn = "10"
input_player1_r_btn = "11"
input_player1_l2_btn = "8"
input_player1_r2_btn = "9"
input_player1_l3_btn = "1"
input_player1_r3_btn = "2"
input_player1_start_btn = "3"
input_player1_select_btn = "0"
input_player1_left_btn = "7"
input_player1_up_btn = "4"
input_player1_right_btn = "5"
input_player1_down_btn = "6"
input_player1_l_x_plus_axis = "+0"
input_player1_l_y_plus_axis = "+1"
input_player1_l_x_minus_axis = "-0"
input_player1_l_y_minus_axis = "-1"
input_player1_r_x_plus_axis = "+2"
input_player1_r_y_plus_axis = "+3"
input_player1_r_x_minus_axis = "-2"
input_player1_r_x_minus_axis = "-3"
input_enable_hotkey_btn = "16"
input_exit_emulator_btn = "16"

joolswills, 2015-03-03:
thanks for the feedback.

closed 2015-03-03 by joolswills