RetroPie/RetroPie-Setup #708: cave story segfaults on launch (Rpi 2)

cacophony555, 2015-03-02:
I have the 2.6 image running on my RP 2, and used retropie_setup.sh to install the cave story core. But when I try to launch it from emulationstation it immediately causes a segmentation fault and quits back to emulationstation. 

I can try to get more logging info for you if needed, but I'm hoping it's easily reproducible and not needed given that I'm basically using an off the shelf 2.6 image.

joolswills, 2015-03-02:
did you install the datafiles ? we don't ship with them now - as it's against the authors wishes. You need /opt/retropie/libretrocores/cavestory/datafiles/Doukutsu.exe

joolswills, 2015-03-02:
(also it is in experimental section - and i've not tested it recently).

cacophony555, 2015-03-04:
I had no idea I needed to install datafiles? Is this documented somewhere? Where can I get these files?

joolswills, 2015-03-04:
It's not documented, and in the same way regarding copyrighted roms, I can't help with that I'm afraid apart from the info I've given - you need the cavestory datafiles in the location above.

closed 2015-03-21 by joolswills