RetroPie/RetroPie-Setup #709: .srm save files end up in the wrong rom directory

cacophony555, 2015-03-02:
The .srm save files for PC Engine games keep ending up in other rom directories for some reason. It’s 100% reproducible.

I’ll start playing Bonk’s Adventure for PC Engine, then I’ll quit the game, exit emulation station, and the .srm file for Bonk’s Adventure will be in the nes-nestopia directory … or the snes directory. I’ll delete the file and launch the game again and the same thing happens. The .srm file never ends up in the pcengine rom directory (where the pc engine roms reside).

As far as I can tell this only happens for PC Engine games as my .srm files for snes games are all in the snes folder.

I’m using the latest version 2.6 of retropie.

joolswills, 2015-03-03:
please can you paste the contents of /opt/retropie/configs/pcengine/retroarch.cfg

cacophony555, 2015-03-04:
Sure, it's blank:

# All settings made here will override the global settings for the current emulator core

Also, if it helps here's the relevant part of es_systems.cfg:

  <system>
    <fullname>TurboGrafx 16 (PC Engine)</fullname>
    <name>pcengine</name>
    <path>~/RetroPie/roms/pcengine</path>
    <extension>.pce .PCE .zip .ZIP</extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 1 "/opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/mednafenpcefast/mednafen_pce_fast_libretro.so --config /opt/retropie/configs/all/retroarch.cfg --appendconfig /opt/retropie/configs/pcengine/retroarch.cfg %ROM%" "mednafenpcefast"</command>
    <platform>pcengine</platform>
    <theme>pcengine</theme>
  </system>

joolswills, 2015-03-04:
you need to update retropie-setup.sh and the emulator in question (from menu 5). There were changes a week or so ago that set the save locations. to get this fixed for all libretro emulators, you should probably do a full binary install (backing up any custom configs you have now just in case).

joolswills, 2015-03-21:
Do you know if this problem is still happening (after doing a full update etc?)

cacophony555, 2015-03-23:
I just verified that this problem seems to be fixed. Thanks for the follow-up.

closed 2015-03-23 by cacophony555