RetroPie/RetroPie-Setup #711: bashwelcometweak doesn't work for wireless

joolswills, 2015-03-03:
http://blog.petrockblock.com/forums/topic/ip-adress-in-startscreen-blank/#post-89973

at the rertropie start screen in terminal, the ip address is empty… May be it is because i use the wlan and my ethernet is disabled.

jens-b, 2015-03-03:
I found out whats wrong, the script searches for “inet addr:”. In german there is no "inet addr:" only “inet Adresse:”.
After i corrected it from “inet addr:” to “net Adresse:” it works.
kind regards
Jens

joolswills, 2015-03-03:
thanks yeh, the script isn't too great in places and shouldn't be matching english strings really. I will sort it when I get a chance.

jens-b, 2015-03-03:
May be if the script only searches for "inet" it should work for most of the languages?

jens-b, 2015-03-03:
hm, think the  headline for this issue has to corrected from "dosnt work for wireless" to "dosnt work in different languages" ?

joolswills, 2015-03-03:
I'm considering using "ip route" which should be more reliable.

joolswills, 2015-03-03:
hopefully fixed now. please try

```
cd
cd RetroPie-Setup
git pull
sudo ./retropie_packages.sh bashwelcometweak
```

and then logout/log back in and see if it sorts it. thanks for the feedback.

jens-b, 2015-03-03:
Hello joolswills,
many thx it works :-).... BTW is that you on the pic? With an old C64? Nice one :-)

kind regards
Jens

joolswills, 2015-03-03:
actually, is my daughter - taken about  5 1/2 years ago :)

Picture was taken on a Holiday in Cornwall. Yes, I know, taking my C64 on holiday.. nerd! Mind you, it usually rains so.

joolswills, 2015-03-03:
thanks for reporting the issue. closing

closed 2015-03-03 by joolswills