RetroPie/RetroPie-Setup #713: Gpsp-Libreto does not work.

expandables2, 2015-03-03:
Hi i tried Gpsp Libreto and it doesn't play any rom. It leaves a GBA.GBA file when i launch a gba rom.

Mutex-, 2015-03-05:
Is there a gba_bios.bin inside your BIOS folder? (~/RetroPie/BIOS)

expandables2, 2015-03-05:
Yes but it still doesn't work. The emulator version of gpsp works but not Libreto. I tried vab-next and that worked also.

joolswills, 2015-03-14:
It works here - please give more details - such as platform / game etc

expandables2, 2015-03-14:
I am using raspbian and i compile gpsp using your script for retropie. I put the gba_bios in retroarch cores where gpsp libretro is but i am getting a GBA.GBA file when i try to launch a game in emulationstation.

joolswills, 2015-03-14:
rpi1 or rpi2 ? have you updated to the latest version of lr-gpsp? (might be worth doing a full binary install to get everything up to date including all the new emulators)

expandables2, 2015-03-14:
I got raspberry pi 2. I am saying i have Raspbian and emulationstation install and i use your script lr-gpsp. I got gpsp-libretro.so and i put it in Retroarch core and i also put gba_bios in retroarch core and i put it in the gba roms folder but when i launch a game i get a file called GBA.GBA. Do you launch gpsp to launch the bios too? and my launcher looks like this COMMAND=~/.emulationstation/runcommand.sh 2 "~/.Emulators/Retroarch/retroarch -L ~/.Emulators/Retroarch/cores/gpsp_libretro.so  --config ~/.Emulators/Retroarch/retroarch.cfg --appendconfig ~/.Emulators/Retroarch/configs/gba/retroarch.cfg %ROM%"

joolswills, 2015-03-14:
Those paths are not used by retropie-setup - this is a custom set up ?

the bios should go to where system_directory points to in retroarch.cfg

if you install retropie-setup from the full binary install with the latest version of the script, it should set everything up correctly.

expandables2, 2015-03-14:
Ok thanks appreciate the help i just setup  system_directory and it works :)

joolswills, 2015-03-14:
:) glad you have it sorted.

closed 2015-03-14 by joolswills