RetroPie/RetroPie-Setup #715: Updating retropie setup script gives error: The following untracked working tree files would be overwritten by merge:

jens-b, 2015-03-04:
Updating 3e2524b..2204ef7
error: The following untracked working tree files would be overwritten by merge:
	scriptmodules/admin/setup.sh
	scriptmodules/libretrocores/snes9x-next.sh
	scriptmodules/supplementary/retropiemenu.sh
Please move or remove them before you can merge.
Aborting

??

joolswills, 2015-03-04:
odd if you haven't changed anything - you didn't add/update with any other method at some point to get the newer files?. you can do as it suggests and delete them if you don't have any changes you need.

gizmo98, 2015-03-04:
Try to stash every change with "sudo git stash" and run retropie_setup.sh again.

petrockblog, 2015-03-13:
I do not see that this is a bug of the RetroPie-Setup Script.

joolswills, 2015-03-13:
it isn't really - more of a support request imho - definitely better suited for the forum. closing.

closed 2015-03-13 by joolswills

scarf, 2017-04-02:
I'm getting the same error trying to update from 4.1 to 4.2, although the files mentioned are different and there's like 30 of them.  I haven't changed any of these files.  See below for file listing.  I started renaming some to .old and it looks like they are timestamped March 19, which might be when I installed 4.1 fresh from an image.  All the files below with a stamp of "Mar 19" are the ones mentioned in the error.  Running git stash from Retropie-Setup dir returns "No local changes to save"

```
pi@retropie:~/RetroPie-Setup $ ls -laR
.:
total 96
drwxr-xr-x 8 pi pi  4096 Apr  2 01:00 .
drwxr-xr-x 6 pi pi  4096 Apr  2 01:09 ..
-rw-r--r-- 1 pi pi   146 Nov  5 21:01 AUTHORS
-rw-r--r-- 1 pi pi   421 Nov  5 21:01 CONTRIBUTING.md
-rw-r--r-- 1 pi pi   635 Nov  5 21:01 COPYRIGHT.md
drwxr-xr-x 8 pi pi  4096 Apr  2 01:09 .git
drwxr-xr-x 2 pi pi  4096 Nov  5 21:01 .github
-rw-r--r-- 1 pi pi   155 Nov  5 21:01 .gitignore
-rw-r--r-- 1 pi pi 32265 Nov  5 21:01 LICENSE.md
drwxr-xr-x 2 pi pi  4096 Apr  2 01:05 logs
-rw-r--r-- 1 pi pi  3797 Nov  5 21:01 platforms.cfg
-rw-r--r-- 1 pi pi  1925 Nov  5 21:01 README.md
-rwxr-xr-x 1 pi pi  1713 Nov  5 21:01 retropie_packages.sh
-rwxr-xr-x 1 pi pi   796 Nov  5 21:01 retropie_setup.sh
drwxr-xr-x 7 pi pi  4096 Apr  2 01:11 scriptmodules
drwxr-xr-x 3 pi pi  4096 Mar 19 19:04 tmp
drwxr-xr-x 3 pi pi  4096 Nov  5 21:01 tools

./.git:
total 80
drwxr-xr-x 8 pi pi  4096 Apr  2 01:09 .
drwxr-xr-x 8 pi pi  4096 Apr  2 01:00 ..
drwxr-xr-x 2 pi pi  4096 Nov  5 21:01 branches
-rwxr--r-- 1 pi pi   271 Nov  5 21:01 config
-rw-r--r-- 1 pi pi    73 Nov  5 21:01 description
-rw-r--r-- 1 pi pi   228 Apr  2 01:09 FETCH_HEAD
-rw-r--r-- 1 pi pi    23 Nov  5 21:01 HEAD
drwxr-xr-x 2 pi pi  4096 Nov  5 21:01 hooks
-rw-r--r-- 1 pi pi 23704 Apr  2 01:08 index
drwxr-xr-x 2 pi pi  4096 Nov  5 21:01 info
drwxr-xr-x 3 pi pi  4096 Nov  5 21:01 logs
drwxr-xr-x 4 pi pi  4096 Nov  5 21:01 objects
-rw-r--r-- 1 pi pi    41 Apr  2 00:49 ORIG_HEAD
-rw-r--r-- 1 pi pi  1533 Nov  5 21:01 packed-refs
drwxr-xr-x 5 pi pi  4096 Nov  5 21:01 refs

./.git/branches:
total 8
drwxr-xr-x 2 pi pi 4096 Nov  5 21:01 .
drwxr-xr-x 8 pi pi 4096 Apr  2 01:09 ..

./.git/hooks:
total 48
drwxr-xr-x 2 pi pi 4096 Nov  5 21:01 .
drwxr-xr-x 8 pi pi 4096 Apr  2 01:09 ..
-rwxr-xr-x 1 pi pi  452 Nov  5 21:01 applypatch-msg.sample
-rwxr-xr-x 1 pi pi  896 Nov  5 21:01 commit-msg.sample
-rwxr-xr-x 1 pi pi  189 Nov  5 21:01 post-update.sample
-rwxr-xr-x 1 pi pi  398 Nov  5 21:01 pre-applypatch.sample
-rwxr-xr-x 1 pi pi 1642 Nov  5 21:01 pre-commit.sample
-rwxr-xr-x 1 pi pi 1239 Nov  5 21:01 prepare-commit-msg.sample
-rwxr-xr-x 1 pi pi 1352 Nov  5 21:01 pre-push.sample
-rwxr-xr-x 1 pi pi 4898 Nov  5 21:01 pre-rebase.sample
-rwxr-xr-x 1 pi pi 3611 Nov  5 21:01 update.sample

./.git/info:
total 12
drwxr-xr-x 2 pi pi 4096 Nov  5 21:01 .
drwxr-xr-x 8 pi pi 4096 Apr  2 01:09 ..
-rw-r--r-- 1 pi pi  240 Nov  5 21:01 exclude

./.git/logs:
total 16
drwxr-xr-x 3 pi pi 4096 Nov  5 21:01 .
[... 506 lines cut ...]

./tmp/build:
total 8
drwxr-xr-x 2 pi pi 4096 Mar 19 19:15 .
drwxr-xr-x 3 pi pi 4096 Mar 19 19:04 ..

./tools:
total 16
drwxr-xr-x 3 pi pi 4096 Nov  5 21:01 .
drwxr-xr-x 8 pi pi 4096 Apr  2 01:00 ..
-rwxr-xr-x 1 pi pi   50 Nov  5 21:01 createMovieFromImages.sh
drwxr-xr-x 2 pi pi 4096 Nov  5 21:01 sdcard

./tools/sdcard:
total 24
drwxr-xr-x 2 pi pi 4096 Nov  5 21:01 .
drwxr-xr-x 3 pi pi 4096 Nov  5 21:01 ..
-rwxr-xr-x 1 pi pi  936 Nov  5 21:01 chroot.sh
-rwxr-xr-x 1 pi pi  807 Nov  5 21:01 flashsync.sh
-rwxr-xr-x 1 pi pi  407 Nov  5 21:01 mount.sh
-rw-r--r-- 1 pi pi  956 Nov  5 21:01 readme.txt
pi@retropie:~/RetroPie-Setup $ git stash
No local changes to save

```

joolswills, 2017-04-02:
Please use the RetroPie forum for support. Looks like you just added or modified some scripts that are part of retropie-setup. `git reset --hard; git clean -f -d` should sort it if run from the RetroPie-setup sir. It will wipe any customisations to the code