RetroPie/RetroPie-Setup #716: optimization for multi-core pi2

ric96, 2015-03-05:
Has the code been optimized to take the advantage of the quad core processor of the pi2 at both compilation and run time ?

joolswills, 2015-03-05:
regarding compilation - see 

https://github.com/petrockblog/RetroPie-Setup/issues/663 and
https://github.com/petrockblog/RetroPie-Setup/issues/691

Some emulators have some simple multi core support such as a separate thread for rendering etc, but most emulators are single threaded. There is no magical way to make something like an emulator use multiple cores, and by design, it is difficult to emulate a system with multiple threads.

ric96, 2015-03-05:
What about compilation of retropie?

joolswills, 2015-03-05:
I just linked you to two previous tickets on the topic, both with the answer.

ric96, 2015-03-05:
Oh yeh, thnx will try it out!!

joolswills, 2015-03-05:
np.

ric96, 2015-03-06:
Neogeo failed to build :(

joolswills, 2015-03-06:
please post the build log (from logs/)

note that some emulators may fail to build with -jX - hence it is not switched on by default. Not all emulators have makefiles/dependencies that are compatible with parallel building.

ric96, 2015-03-06:
Gamegear too failed

ric96, 2015-03-06:
I'll post it once it's over,
I did want to log whatever fails on -j4 so that it can be fixed or maybe an exception could be added so that they are always built without -j4

joolswills, 2015-03-06:
It's not always consistent, but that is already the plan - however I am busy on other things currently. Did you read the other threads regarding this as I went into some detail.

ric96, 2015-03-07:
log: https://drive.google.com/file/d/0B4VR-WNTOX0qNm9abm5qd0kxYUU/view?usp=sharing

also on starting emulationstation it says:

ric96, 2015-03-07:
https://drive.google.com/file/d/0B4VR-WNTOX0qQVRTdVpJQUdZNnc/view?usp=sharing

gizmo98, 2015-03-07:
If i run emulationstation with sudo or as a different user i have the same message. But this is not the normal use case.

ric96, 2015-03-07:
If I don't run with sudo it gives me permission denied error!!

joolswills, 2015-03-07:
yeh from the log osmose/gngeo failed. this is usual. for those you can build without -jX for now.

joolswills, 2015-03-07:
you get permissions denied as ~/.pi/.emulation station is now owned by root. you need to chown it. But just install from binaries and everything is simpler :)

joolswills, 2015-03-07:
chown -R pi:pi ~/.emulationstation

closed 2015-03-13 by joolswills