RetroPie/RetroPie-Setup #719: Automatic configurations across all emulators for wireless PS3 controller support

lip5016, 2015-03-06:
Raspberry Pi 2
RetroPie 2.6.0

After experiencing it to great lengths myself, I think the most frustrating thing after getting a brand-new Raspberry Pi 2 all set up with RetroPie 2.6.0, is the task of connecting and configuring button mappings for *2 or more* wireless/bluetooth Playstation 3 PS3 controllers.

There is so much dated and misleading advice when searching on Google/YouTube, that it really makes it tough for a beginner (which I'm sure accounts for 99% of people who buy a RPi2, just so they can jump right in to RetroPie). 

So many websites, YouTube videos, blogs, Reddit, etc. offer "advice", but since the RetroPie project moves so quickly and is constantly growing and changing, much of the advice that is out there becomes extremely daunting and confusing; making an inexperienced user overwhelmed with what is actually current/necessary information, and what is outdated/already implemented information. 

I feel like an "automatic PS3 controller configuration" would eliminate the need for new users to go crazy with trying to figure out how to edit all types of .cfg (config) files for each emulator on RetroPie. Some people want to have a Raspberry Pi 2 because they're super into programming and command-line, but there are others out there like myself, who purely just want a 1-2-3, load it up and go retro gaming console. 

I don't buy the argument "one size doesn't fit all" when it comes to mapping controllers to each specific emulator. Video game controllers are simple. Most people just want a functioning controller, which has the most similar dynamic for which buttons are which. A SNES Super Nintendo controller should only be mapped 1 single way when setting it up on a PS3 controller. If you feel the need to be the guy who has to have unconventional button layout schemes on their gaming consoles, you could always go back and edit the configs yourself, but for everyone else, a "one size fits all" wireless PS3 configuration is probably more than enough.

tl;dr
someone please write something that can allow a user to simply press the PS button on their PS3 controllers, and automatically be connected wirelessly via bluetooth, with all the RetroPie emulators all configured automatically across all the emulators on RetroPie

joolswills, 2015-03-06:
There is ongoing work for ps3 controllers currently (check earlier tickets), but I would point out as a dev, I don't own a ps3 controller.

It is a lot of work to automatically configure every single emulator as many use various button schemes / mapping - would you be willing to contribute towards this ?

joolswills, 2015-03-06:
I would add, that there are lots of available controller out there, and so getting every one working out of the box isn't always possible, but we work towards that - by adding user contributed controller configs for retroarch etc. in terms of having a controller work, if you stick to retroarch emulators, once you have paired your ps3 controller it should work. Hopefully connecting/pairing etc will be improved with upcoming releases.

lip5016, 2015-03-06:
I would love to. But I'm literally experiencing the same problem I'm posting about. All these different .cfg's and editing files is overwhelming. I'd be here to do whatever beginner tasks you guys could afford to assign to me, but besides that, I'm afraid I wouldn't be much help :(

joolswills, 2015-03-06:
we were all beginners once ;-) if you stick to retroarch/librerto emulators you shouldnt have to edit lots of config files. apart from that, I would recommend getting help from the forum for now.

lip5016, 2015-03-06:
That's what I plan! I actually had to reach out to petrockblog on Reddit because for weeks, his forum wouldn't let me register :( They have some crappy security company implementation that keeps saying I'm banned because it's mistaken me as a spammer for some reason? Either way, it's all sorted, as I've never actually even been a member of the forum to ever warrant getting banned (lol). Petrockblog agreed and yesterday said he would manually register me with my e-mail address, so now I'm just patiently waiting with excitement =D

lip5016, 2015-03-06:
Also, I originally wanted to go the route of all the standard emulators (those simply named, "nes", "snes", "n64", etc.), but there are too many options now IMO! and it seems like the "best" emulators are the ones with the suffixes at the end of their folder names (for instance, n64-mupen64plus seems to run n64 games way better than the simply named "n64" folder).. and the headache comes into play when you realize that the mupen64plus option is the best one, and also the same option that doesn't come with all the controller configs worked out :(

joolswills, 2015-03-06:
there are too many options, and I hope to have a solution to this soon, so you have a single rom folder, but can switch which emulator is used. we will also change which are the defaults etc, and get feedback from the users on this.

lip5016, 2015-03-06:
That would be awesome. I can understand why there might be several different emulators for each category implemented. Since certain emulators will run certain games better, but there has to be a more simple way. I've got several thousand ROMs and would just love to be able to drop them onto the RetroPie and go, without any sort of constant tinkering, etc. Hopeful, I know, but with the explosion of the Raspberry Pi 2 release, I feel like there are enough passionate people out there who understand code and command-line enough to make this a reality.

jedahan, 2015-03-06:
I would love for something as simple as a script be distributed per-rom in the following format.
For example, for zelda.n64, a file zelda.sh, with the following contents:

```bash
#!/usr/bin/env bash
mupen64 -v ricevideo -g zelda.n64 -nogui -f
```

Then, an option in retropie to prefer the .sh file extensions or to ignore them.

The nice thing about the shell scripts, is that people can do really flexible stuff with them, for example, we could have a few default scripts like `mupen-rice.sh` and `mupen-glide.sh` and all you would need is a symlink for all the roms, with `ln -s RetroPie/contrib/launchers/mupen-rice.sh roms/n64/zelda.sh` and thats it.

We could build a decent database of launchers as a separate git repository, and RetroPie-setup can just pull in whatever we think is the best audited one.

Any thoughts?

robertybob, 2015-03-06:
@jedahan Not sure if you're aware of the Dolphin GC/Wii emulator, but they do something similar with .ini files which correspond to every possible game ISO (it runs into the 1000s obviously). 

Basically the software reads the ROM and loads the corresponding .ini file to enable whichever configuration suits the game best, obviously that config has to be determined by someone, but it generally works well.

I feel, for a system like N64 where there aren't hundreds of ROMs out there, that this could work as it wouldn't be such a massive job to test every game (compared to the SNES) and decide on the best configuration for the RPi2.

jedahan, 2015-03-06:
Yeah it depends on who wants to support it - ini files work well if the emulator developer wants to package settings files, but what happens if a certain n64 game works better on mupen and another works better on a different emulator entirely? Then the ini format has to be coordinated between the emulator developers, which would take valuable time away from them working on the main emulation.

Also, by making the scripts external to the emulators and even emulator frontend, we can accept much more community scripts and development, kinda like [homebrew](http://brew.sh)...

I thought of an ini format first, but the flexibility of using scripts to launch stuff just seems so appealing to me - no dependencies, no need to integrate code upstream...

joolswills, 2015-07-13:
We now have an integrated configuration - so after configuring a controller in emulationstation it now configures retroarch also. We will add additional scripts to manage non retroarch emulators in the future (such as pifba).

closed 2015-07-13 by joolswills