RetroPie/RetroPie-Setup #721: [Enhancent] Integration of Limelight Streamer

jayp76, 2015-03-07:
Is it possible to integrate Limeligt?

joolswills, 2015-03-08:
looks interesting. adding this link for reference https://github.com/irtimmer/limelight-embedded

stsfin, 2015-03-26:
I made a Guide to the wiki how to manually add it to retropie and emulation station: https://github.com/petrockblog/RetroPie-Setup/wiki/Quickguide-for-configuring-limelight-with-Retropie-and-Emulation-Station

just updated the wikipage after I found working OC settings to get the 1080p/60FPS streaming running... 1080p/30fps and 720p/60fps work without OC.

stsfin, 2015-03-27:
I'll probably try to make an script that automates the install as much as possible (more on that later)

stsfin, 2015-03-27:
Here is the installer script I created:

https://github.com/stsfin/RetropieLimelightInstaller/blob/master/limelightRetroInstall.sh

Should help integration.

stsfin, 2015-03-27:
Updated the installer script with eventID selection. Now the controller can be in anyport. I will create a secondary script with only ip and controller reconfiguration to make it easier

stsfin, 2015-03-27:
Updated the script, it is basically now complete, tested with 3 different retropie installations: https://github.com/stsfin/RetropieLimelightInstaller  
@joolswills be free to fork it for RetroPie :)

stsfin, 2015-03-29:
Im having trouble in integrating the reconfig script to Emulation station.  The read command just gives "ead: read error: 0: bad file descriptor" when running the reconfig-script from the ports-section... it works nicely from the console... @joolswills any pointers?

stsfin, 2015-03-29:
The installer now creates a new "device"-menu under Emulation station for better integration... Still no luck with the reconfiguration script though.

stsfin, 2015-03-29:
[image]
[image]
[image]

joolswills, 2015-03-29:
regarding the problem - use `<dev/tty` on the read command.

stsfin, 2015-03-29:
@joolswills , thanks will try it out if it works i will update the script to add the reconfigurator to the menu

stsfin, 2015-03-29:
Thanks for @joolswills for helping me with the last obstacle... Here it is, the final version of the script with the reconfigurator accesible from emulation station after install: https://github.com/stsfin/RetropieLimelightInstaller/archive/2.0.1.zip

stsfin, 2015-03-29:
Here is how it looks and works after install https://youtu.be/CT0Zhmh9e9o

taalas, 2015-04-20:
Commented on some minor issues in the merged code. Should I rather have commented here?

stsfin, 2015-04-20:
If you are refering to the code that will be implemented to Retropie you
should comment in that thread.

Jonne Backhaus
______________
+358-45-1141905
jonne.backhaus@gmail.com
jonne.backhaus@androidsuomi.fi

Google+: http://backhaus.fi/+/
LinkedIn: http://backhaus.fi/linkedin/

taalas, 2015-04-20:
I commented in Rewrite limelight setup script to a module #792

closed 2015-06-20 by joolswills