RetroPie/RetroPie-Setup #725: Github organization "RetroPie"?

petrockblog, 2015-03-09:
Do you think that we should open a Github organization "RetroPie"? Would we have any benefits? Or would we just create unnecessary overhead? I would need to change quite a few URLs and the new URL would not be so well known at the beginning.

@joolswills @gizmo98 What do you think?

joolswills, 2015-03-09:
Could be useful to keep all the repositories in one place at least, but it's not absolutely required if you don't think it's needed. I'm not in any rush for it - far too much ongoing stuff currently to worry about updating urls :) but perhaps when/if things settle down.

gizmo98, 2015-03-10:
A totally agree with jools. If there is less ongoing stuff we can think about "one" place for all retropie repos.

petrockblog, 2015-04-06:
I have reserved retropie as organization at Github. We'll see, if we need this at some point on time ...

closed 2015-04-06 by petrockblog

joolswills, 2015-06-01:
I think it would be a good time to do this - are you happy to move this and the other related repositories to the organisation ? If you make me an admin I can do the same for my repositories (and gizmo can do for his etc).

petrockblog, 2015-06-01:
Sure - I will do that tomorrow!

joolswills, 2015-06-01:
cheers!

petrockblog, 2015-06-15:
I have finally moved the repository to the organization. @joolswills and @gizmo98, I have send you two an invitation to become team members. Still have to figure out the best access rights settings, though.