RetroPie/RetroPie-Setup #726: [Feature Reqest] USB copy Gamelist and Downloaded_Images

robertybob, 2015-03-10:
I like the USB romservice feature but I don't understand why we don't do the same thing for gamelist.xmls and the scraped/downloaded images/box arts we have. I create all of these on my PC and rather than keep using SSH to transfer them over, I'd like to be able to just plug my USB stick into my Pi, my PC, then back into my Pi and have it copy over all of my ROMs, artwork and custom gamelist.xml files in one go.

petrockblog, 2015-03-13:
In addition it would make sense to also sync configuration files with the USB stick. In this way the USB copy service could become a tool for a simple backup and restore of most of the customized things of the users.

joolswills, 2015-03-13:
how would you envisage the configurations to be synced ? I'm just thinking whether it would be confusing if roms are synced one way, but configurations the other.

petrockblog, 2015-03-14:
Good point.
Maybe it would be a good compromise, if there were the folders `configs_fromRetroPie` and `configs_toRetroPie`.
After mounting the USB service would create a new backup of the configuration files (maybe in an archive, keeping older revisions of configurations in separate files).
Per default the folder `configs_toRetroPie` contains only a `read.txt` telling the user that if he puts configuration files in that folder they will be copied over to the RPi. 
A backup and restore use case would then consist of
1. There user hooks up the USB stick to the RPi, the USB service copies all configuration files to the folder `configs_fromRetroPie`.
2. The user copies the configuration archive to the folder `configs_toRetroPie`.
3. The user hooks up the USB stick to the RPi, the USB service sees that there are configurations to import and extracts/copies these configurations to their corresponding folders.

These are brainstorming thoughts, though. I might have overlooked something.

petrockblog, 2015-03-15:
I have committed what I have basically described above (see commit d0c798417264426394ce68c334a8de537865fd14). However, before we close this issue we need to discuss, which set of data actually belongs to the configurations that we want to have a backup of. The [copy script](https://github.com/petrockblog/RetroPie-Setup/blob/master/supplementary/01_retropie_copyroms) needs to be adapted accordingly then.

joolswills, 2015-03-17:
I think if doing this we may want to make the changes discussed in https://github.com/petrockblog/RetroPie-Setup/issues/619 .

I also think we should have three usbromservice options - install with copy script, just install service, and remove (we have install/remove currently afair). Possibly making the config backups optional.

It's worth noting that there will be dangers involves in restoring configs automatically as some stuff changes that is not backward compatible, so restoring old configs over a new version could cause issues. we should at least warn about this perhaps, so people are aware.

joolswills, 2015-03-17:
I made a few minor tweaks to the script, but ill submit a pull request with some additional changes - to allow easy addition of new config files/paths

joolswills, 2015-03-17:
Just realised you already handle the case to not sync if the usb doesn't have a retropie folder. so that solves the other ticket.

robertybob, 2015-04-07:
Since 3.0 has been released, I'm making the assumption this can be closed now :)

closed 2015-04-07 by robertybob