RetroPie/RetroPie-Setup #729: Modify N64 InputAutoCfg.ini if a controller is registered

gizmo98, 2015-03-12:
If a controller is registered we can use the retroarch autoconfig file to modify /opt/retropie/configs/n64/InputAutoConfig.ini. Most button numbers are equal. AnalogDeadzone and AnalogPeak is always the same. Retroarchs input_device is the section name of InputAutoConfig.ini.

Example:
https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/PS3Controller.cfg
input_device = "PLAYSTATION(R)3 Controller"
input_driver = "udev"
input_b_btn = "14"
input_y_btn = "15"
input_select_btn = "0"
input_start_btn = "3"
input_up_btn = "4"
input_down_btn = "6"
input_left_btn = "7"
input_right_btn = "5"
input_a_btn = "13"
input_x_btn = "12"
input_l_btn = "10"
input_r_btn = "11"
input_l2_btn = "8"
input_r2_btn = "9"
input_l_x_plus_axis = "+0"
input_l_x_minus_axis = "-0"
input_l_y_plus_axis = "+1"
input_l_y_minus_axis = "-1"
input_r_x_plus_axis = "+2"
input_r_x_minus_axis = "-2"
input_r_y_plus_axis = "+3"
input_r_y_minus_axis = "-3"

https://github.com/mupen64plus/mupen64plus-input-sdl/blob/master/data/InputAutoCfg.ini
[Sony PLAYSTATION(R)3 Controller]
plugged = True
plugin = 2
mouse = False
AnalogDeadzone = 4096,4096
AnalogPeak = 32768,32768
DPad R = button(5)
DPad L = button(7)
DPad D = button(6)
DPad U = button(4)
Start = button(3)
Z Trig = button(10)
B Button = button(15)
A Button = button(14)
C Button R = axis(2+)
C Button L = axis(2-)
C Button D = axis(3+)
C Button U = axis(3-)
R Trig = button(11)
L Trig = button(8)
Mempak switch = 
Rumblepak switch = 
X Axis = axis(0-,0+)
Y Axis = axis(1-,1+)

gizmo98, 2015-11-17:
I will try to create a mupen64plus config script.

closed 2015-11-17 by gizmo98

HerbFargus, 2015-11-17:
I'm happy to test once you come up with something

HerbFargus, 2015-11-18:
I try to find time this evening to give it a go. will hotkeys also be autogenerated with this or is that a separate thing?

gizmo98, 2015-11-19:
Copy files to... 
/opt/retropie/supplementary/emulationstation/inputconfiguration.sh
/opt/retropie/supplementary/emulationstation/configscripts/mupen64plus.sh

I frame new configs with CONFIG START/CONFIG STOP markers. If you try a reconfig, current config will only be replaced if i find these markers for a certain controller. So original controller configs of repo mupen64plus-input-sdl will not be overwritten. 

Delete /opt/retropie/config/n64/InputAutoConfig.ini if you want to test with a clean ini file.

HerbFargus, 2015-11-19:
Brilliant mate :) 

here is my cheap snes knock off controller config that was generated:

```
; InputAutoCfg.ini for Mupen64Plus SDL Input plugin

[Keyboard]
plugged = True
plugin = 2
mouse = False
DPad R = key(100)
DPad L = key(97)
DPad D = key(115)
DPad U = key(119)
Start = key(13)
Z Trig = key(122)
B Button = key(306)
A Button = key(304)
C Button R = key(108)
C Button L = key(106)
C Button D = key(107)
C Button U = key(105)
R Trig = key(99)
L Trig = key(120)
Mempak switch = key(44)
Rumblepak switch = key(46)
X Axis = key(276,275)
Y Axis = key(273,274)

;USB gamepad            CONFIG START
[USB gamepad           ]
plugged = True
plugin = 2
mouse = False
AnalogDeadzone = 4096,4096
AnalogPeak = 32768,32768
Mempak switch = 
Rumblepak switch = 
R Trig = button(5)
Start = button(9)
Z Trig = button(4)
DPad U = 
A Button = button(2)
DPad D = 
DPad R = 
B Button = button(3)
DPad L = 
Y Axis = axis(1-,1+)
X Axis = axis(0-,0+)
;USB gamepad            CONFIG END
```
all thats left is to autogenerate the hotkeys ;)

It was a little unintuitive to me at first as the controls were mapped physically based on how the original n64 was rather than the default core controls that retroarch sets (see https://cloud.githubusercontent.com/assets/10035308/7334793/6b406d8c-eb5b-11e4-8474-56a340521e71.png) Instead of B A like retroarch, my controller was mapped Y B which is actually closer to how the original n64 was (not sure if you did that intentionally or not)

gizmo98, 2015-11-19:
I used Ps3 controller mapping of input-sdl repo as a reference for A B. So it's intended.

Your F310 config should not be overwritten. I had no problem like this with my controllers. I will take a look later.

gizmo98, 2015-11-19:
Found it. I will add missing path variable to line 219. ${path}${file}

gizmo98, 2015-11-19:
Do you think mapping C Down and C Left to button A/X is a good idea?

HerbFargus, 2015-11-19:
On the snes controller? If they aren't being used, may as well - I honestly don't play the n64 enough to know whether it would be beneficial or not, if someone has a right analog would you just have duplicate mappings for the C buttons?

gizmo98, 2015-11-19:
Duplicate mapping is no problem. Mupen64plus supports multimapping.

For snes controllers I mapped C button DOWN/UP to X/A, so i can play zelda with navi and one item. (Ok, ocarina needs all c buttons. So this is a nogo). Without these buttons you cannot play this game at all.

I changed the script a little bit. Could you please test again?

HerbFargus, 2015-11-19:
Yeah I'll give it a go once I get off work in ~ 4 hrs

HerbFargus, 2015-11-20:
Works great :) For reference this is my config for both controllers:

http://pastebin.com/yGn1K8rB

At some point it might be worth integrating hotkeys with what jools referenced here:

https://github.com/RetroPie/EmulationStation/issues/12

gizmo98, 2015-11-20:
Thank you very much! 

For hotkeys we still need an extra package (https://github.com/RetroPie/RetroPie-Setup/issues/1061). Mupen64plus handles hotkeys and default input independent. I don't think mupen64plus has issue like retroarch because it uses an sdl event filter to detect a hotkey press. No button will be blocked. If there will be a option to for "enable hotkey button" i would like to change this package so hotkey behavior is consistent for all emus.

HerbFargus, 2015-11-20:
I agree, a consistent behaviour would be best- there was some progress on the hotkey blocking issue

https://github.com/libretro/RetroArch/issues/2230

But either I was testing wrong or the code was buggy as my results were sporadic, perhaps you'd have a better understanding of it. 

Also as far as the n64 hotkey function you set up are the udev rules necessary or are you hoping to make those obsolete with a centralised hotkey function?

gizmo98, 2015-11-20:
I think a udev rule is the easiest way to configure emus/hotkeys with static bindings to a /dev/input/js* device (gpsp, mupen64plus-core, mame(?)...). I don't think there is a udev rule or centralized hotkey function. In the end a udev rule can be setup to incorporate with a centralized function.