RetroPie/RetroPie-Setup #731: mupen64plus-testing & mupen64plus

joolswills, 2015-03-13:
@gizmo98 

Are there any major differences between these now - looking at the repository info it looks as though mostly it's just the addition of video-glide64mk2 in testing. If so can we please drop the -testing one and just include options to be able to switch video driver as I added at the end of mupen64plus-testing.sh

gizmo98, 2015-03-13:
No major differences anymore:
-glide plugin
-video settings repo
-a list of all test repos

Narann from mupen64plus has done a big video-rice rework which is not ready atm. I would like to use this plugin with -testing until it is ready for the normal module.

joolswills, 2015-03-13:
ok. thanks for the info.

andrewsil1, 2015-03-13:
Just out of curiosity, is there a good reason why this is still using the fork of the core module rather than the main mupen64plus-core repo?  Looks like there just some core commits related to building for RPI2 but it doesn’t look like the fork of core has been updated in quite some time.

joolswills, 2015-03-13:
I don't think it is using the fork of the core module - Where do you see that ?

andrewsil1, 2015-03-13:
Sorry, my bad, I was building the non-experimental mupen64plus
(Saw it syncing to https://github.com/ricrpi/mupen64plus-core)

Rather than building mupen64plus-testing.  Which does use the main branch.  Duh.  Carry on!

_____________________________
Andy Silverman
Principal Program Manager, Microsoft Manufacturing Test Engineering
sendnowmarker

joolswills, 2015-03-13:
is your retropie-setup up to date ? The non-experimental one should also use the main branch.

andrewsil1, 2015-03-13:
Unless this was changed in the last 12 hours or so, which was the last time I did the retropie-setup update (assuming the “get latest” option inside the script itself works correctly), then yeah, it is!
_____________________________
sendnowmarker

joolswills, 2015-03-13:
It has changed in the last 12 hours yes - if you update it will use the core repository.

You using email to reply ? your replies are including the previous message - you probably don't want that - I have stripped the quoted text from the messages.

andrewsil1, 2015-03-13:
You’re right.  Sorry for the false alarm, and also for the note on the attached previous mails. (Work email threads are on by default for me. :-/)  I’ll try to remember to strip ‘em when responding to github from now on.

_____________________________
Andy Silverman

closed 2015-03-14 by joolswills

dankcushions, 2015-08-08:
Hi! I'm trying to make sense of the various mupen64plus versions - is -testing still needed? It looks like mupen64plus core (https://github.com/mupen64plus/mupen64plus-core) and the various video plugins are still getting updates - will these still be pulled in if I rebuild -testing?