RetroPie/RetroPie-Setup #736: PSX Bios

Neuss1987, 2015-03-15:
Hey guys,
I found a page to Download several ps BIOS. After I put them in the BIOS folder and checked /opt/retropie/psx/retroarch.cfg if -> system_directory= /home/pi/RetroPie/BIOS

But after reboot EmulationStation wont show the PS/X.... Menu.

Is there a second config I have to change?
(I use RPi2 with Download-Image)

Benjamin

gizmo98, 2015-03-15:
Put a *.rom/bin/cue in your psx rom directory.

Neuss1987, 2015-03-16:
Hey gizmo,
i added these folders in ..home/RetroPie/roms/psx/.... but it wont work (but im not shure if its the right bios - maybe it'S corrupt - you have a page how to get the right one? 
Do i have to change configs?

agenttank, 2015-03-19:
you have to add the iso/rom/bin/cue to home/RetroPie/roms/psx/...

BIOSes belong to /home/RetroPie/BIOS
GAMES belong to /home/RetroPie/roms

it doesn't matter if there's a bios there, the games should be shown in emulationstation nonetheless.

closed 2015-03-23 by joolswills