RetroPie/RetroPie-Setup #738: mame4all settings do not save in latest 2.6 RetroPie

cacophony555, 2015-03-17:
I started with a fresh RetroPie 2.6 a couple days ago and also did an update to the latest binaries. Everything works great with the exception of mame4all, which doesn’t remember my controller settings.

To be clear I’m talking about the standalone mame4all, not the retroarch core. I’ve tried doing a “chown -R pi:pi” on the mame4all executable directory and confirmed that it changed the ownership of that dir and its contents to user pi. I’ve also verified that the corresponding config directory and contents are owned by pi (though not sure that this dir is even used in this case).

But if I launch mame4all via the Emulation Station or the command line, and then press TAB to edit controller settings, they only persist for that single session.

joolswills, 2015-03-17:
I will look into it. I have a suspicion it's related to our config file moving etc.

biscuits99, 2015-03-17:
Must be a recent change, its fine on my 2.6. (I have not updated any binaries)

joolswills, 2015-03-17:
please update retropie-setup to the latest version, and either re-install from binary via menu 5 (install individual emulators from binary or source), or do

```
sudo ./retropie_packages.sh mam4all configure
```

note if you have any old configs in /opt/retropie/configs/mame4all you can copy them to /opt/retropie/configs/mame-mame4all after doing this step (and then remove /opt/retropie/configs/mame4all)

cacophony555, 2015-03-18:
Thanks! So that did work, but I needed to chown the new config directory to user pi first because the configuration for the controller was still owned by root.

joolswills, 2015-03-18:
that was another bug. should be sorted now.

joolswills, 2015-03-18:
thanks for the feedback.

closed 2015-03-18 by cacophony555