RetroPie/RetroPie-Setup #739: libsdl2-dev dependency

ProxyCell, 2015-03-17:
Please do a quick search through this repo for "libsdl2-dev" and you will find lots of references to it.
Unfortunately I can't seem to install this by just doing "apt-get install libsdl2-dev"
The package system is never able to locate it.

joolswills, 2015-03-17:
it's not in raspbian - it is provided as part of retropie. The script automatically installs it as needed.

Not sure what issue you are referring to - please can you be more specific ?

ProxyCell, 2015-03-17:
oh ok, i just started the gPSP GBA emulator source build and it began downloading it from the petrockblog website

i'm sorry, i thought it was a mistyped dependency there because the same emulator then downloaded libsdl1.2-dev

closed 2015-03-17 by ProxyCell

joolswills, 2015-03-17:
gpsp uses sdl1.2-dev not sdl2-dev.

retropie includes custom sdl1.2 packages and sdl2 packages.

joolswills, 2015-03-17:
(the reason being to fix up some issues with sdl1.2  that used to cause black screens, and also the inclusion of a single library that has a dispmanx backend also)