RetroPie/RetroPie-Setup #741: Audio issues with standalone mame4all

cacophony555, 2015-03-18:
There are pretty significant audio issues with the standalone mame4all that is included with the latest RetroPie 2.6. There is constant crackling, static, and choppy audio with most roms I've tried.
(eg. 1942, Donkey Kong, Ghouls and Ghosts)

None of these audio issues exist with the mame4all binary that is available on the author's website:
https://sourceforge.net/projects/mame4allpi

To be clear I copied the original author's binary to the same Pi 2, and used the same roms. So it seems there must have been a change to the code or how it was built for RetroPie that is causing these problems.

joolswills, 2015-03-18:
I will look into it. I didn't notice naything previously, can you give me an example game to check.

joolswills, 2015-03-18:
note that the only real difference is the retropie build is compiled for the pi2 (if running on pi2) - the original is optimised for the pi1. could you try

```
sudo __platform=rpi1 ./retropie_packages.sh mame4all install_bin
```

and see if that makes any difference (it will install a rpi1 optimised mame4all). it will help me narrow down the possible cause.

joolswills, 2015-03-18:
Also - when you installed the other other binary from sourceforge, did you use our .cfg file or the shipped one ? It might just be that our change to samplerate is the cause, so worth testing editing 

/opt/retropie/emulators/mame4all/mame.cfg

and setting "samplerate" to 44100 to see if that helps. I have it on my todo list to change this anyway, as I noticed some audio lag that seemed to go away when switching to 44100.

cacophony555, 2015-03-18:
Thanks I'll try the two suggestions you mention later today.

To answer your question, when I download the binary from sourceforge I didn't modify anything, so it was using all the default configurations (and as you pointed out it was also not optimized for the Pi 2).

cacophony555, 2015-03-18:
So I just confirmed that changing the samplerate of the mame4all that was installed with RetroPie to 44100 fixes the issue! So I think you figured it out.

joolswills, 2015-03-19:
This is now done. thanks.

closed 2015-03-19 by joolswills