RetroPie/RetroPie-Setup #742: [Game request] Descent 1 & 2!

C-Fu, 2015-03-18:
Somebody have ported/patched the open source version of Descent 1 & 2 for RPi (and RPi 2)!! 

And somebody else have kindly made a script to automate installation of the game on the latest posting on that thread.

http://www.raspberrypi.org/forums/viewtopic.php?f=78&t=14108

I'm playing the game right now on RPi2, almost super smooth, getting 45-55fps on 1080p on max graphic settings :dancers:

MattWatt, 2015-04-07:
Wow, that's really cool.  One of my favorite games growing up.  I'd love to try this on the Pi.

kofronpi, 2015-04-24:
Amazing, I've got to try it !

free5ty1e, 2015-04-27:
I've created a script to install Descent 1 and Descent 2 including the required shareware files.  At some point I want to learn how to turn it into a RetroPie script; in the meantime, if anyone is interested to try it out, here is the script:
https://github.com/free5ty1e/primestationone/blob/master/bin/installDescent1and2.sh

and here are the launch scripts to put in EmulationStation under PORTS:
https://github.com/free5ty1e/primestationone/blob/master/RetroPie/roms/ports/Descent1.sh
https://github.com/free5ty1e/primestationone/blob/master/RetroPie/roms/ports/Descent2.sh

Enjoy!

taalas, 2015-04-28:
Looking forward to trying that. Thanks you!

MattWatt, 2015-06-01:
This is now in the experimental install menu and can be closed.

closed 2015-06-01 by joolswills