RetroPie/RetroPie-Setup #743: Emulationstation

expandables2, 2015-03-19:
Hi good day, How do i uninstall emulationstation?

joolswills, 2015-03-19:
You have given me very little information, but assuming you have it starting on boot:

```
sudo ./retropie_setup.sh
```

go to menu 3 setup - and choose the option to configure emulation station auto starting, and disable on boot.

You can remove the emulationstation binaries by doing

```
rm -rf /opt/retropie/supplementary/emulationstation
```

but you might as well leave them there. Note the retropie modules currently assume /etc/emulationstation/ exists as they configure the menus there, so you should leave that in place - but you could remove the theme files from that folder if you wanted to save some bytes.

expandables2, 2015-03-19:
OK thanks appreciate the help :)

joolswills, 2015-03-19:
welcome.

closed 2015-03-19 by joolswills