RetroPie/RetroPie-Setup #744: Dispmanx Support (native) for Retroarch

amigagamer, 2015-03-20:
Retroarch now has a native dispmanx driver thanks to Vanfanel
 https://github.com/libretro/RetroArch/commit/d7a2a7c4c120fd94b91898ed18ed096005a53596 

adding --enable-dispmanx  to the retroarch ./configure is all thats needed to compile the support ,so it should work from retroarch.sh build_retroarch()

The driver is very new but could be considered as an alternative option, or compiled in for testing purposes.  The GL driver still compiles, so setting video_driver = "gl" in retroarch.cfg should allow retroarch to keep its current configuration.  Dispmanx driver can be enabled with video_driver = "dispmanx" or through the RGUI

gizmo98, 2015-03-20:
Is there any benefit if we add dispmanx? We already have "gl", "sdl" and "vg" and a video core IV dispmanx context driver "videocore".

joolswills, 2015-03-23:
We can enable it perhaps, but I can't see any obvious benefit of using it over the current drivers.

joolswills, 2015-10-10:
Closing for now - happy to reconfigure if someone can argue a valid use case why we need this :)

closed 2015-10-10 by joolswills

julianxhokaxhiu, 2015-11-29:
I'm actually fighting with this issue ( see https://github.com/libretro/RetroArch/issues/2494 ) and it seems that the source code is buggy somewhere and it cannot be really enabled ( at least on Arch Linux ). But when it will work it will definitely solve some tearing problems that we can encounter with sdl2, vg or gl. Don't know if the context of videocore is also having this problem, but dispmanx seems the fastest way to render on RPi.

objectchris, 2015-11-29:
check out this thread - http://blog.petrockblock.com/forums/topic/best-possible-input-lag-with-retroarch/#post-110239 - looks like someone has it working on raspian?

i think this might be worth re-opening as input delay improvements have a real-world benefit? interested to see whether it helps some of the more demanding 2d mame games, also.

joolswills, 2015-11-29:
I doubt it will have a noticeable difference in actual game performance, but I can add dispmanx backend and it can be optionally enabled by users should they want.

joolswills, 2015-12-01:
There was an issue with video_threaded =true but that has been sorted now - https://github.com/libretro/RetroArch/issues/2503 so I will add it as an option.

joolswills, 2015-12-01:
Done. You can switch the driver using the config editor also.