RetroPie/RetroPie-Setup #745: Wiimote not pairing

Firebreath15, 2015-03-20:
After following the tutorial here on GitHub for pairing a Wiimote to a Pi, EmulationStation is still unresponsive when I hold down the 1 + 2 buttons. I have a tested working Bluetooth adapter and a genuine Nintendo Wii remote. I got the MAC address and put it in the files as described, but nothing seems to be working. Is there a known issue with my setup, or am I doing something wrong? I'd appreciate help, preferably with exactly which files to modify/add/delete.

I have a Raspberry Pi 2 Model B at "Modest" overclocking. I installed ES and RetroPie together through this script earlier this week, so everything should be up-to-date. I'm using a Belkin Bluetooth adapter that works just fine, as well as a Netis WiFi dongle. I am using a wireless keyboard, also, but I don't want to use it for ES.

Thank you for your help! If this was posted in the wrong place, then I'd prefer some help via email ;)
Thanks!

joolswills, 2015-03-20:
you might do better to ask on the retropie forums first - http://blog.petrockblock.com/forums/forum/retropie-project-forum/ - to get advise from any users with a similar set-up

Firebreath15, 2015-03-20:
Sounds good. Thanks!

P.S. If you know anything about a "duplicate absolute axis assignment" warning, send it my way ;)

closed 2015-03-20 by Firebreath15