RetroPie/RetroPie-Setup #748: neogeo's emulators.cfg lacks lr-fba

ProxyCell, 2015-03-21:
I was just setting up default emulators on my machine and noticed that neogeo is not even listing lr-fba, only the original (non-lr) fba

Of course the actual FBA emulators.cfg file lists both.

Is there a technical reason for this or has it just not yet been implemented?

ProxyCell, 2015-03-21:
additional: while i know the following could be classed as their own "issue" here I figure they are too closely related

1 - mame - while i have only installed a single "mame" emulator (lr-imame4all) the emulators.cfg does not list any "default"
2 - segaCD - both of their emulators.cfg list only picodrive but I am fairly certain that genesis plus GX can support sega-CD games (it absolutely can not handle 32X, just an FYI)

joolswills, 2015-03-22:
it wasn't implemented. I added lr-fba to the available emulators now - but of course, will require a matching romset - so a romset that works for gngeopi may not work for lr-fba

lr-imame4all is not set as the default on install (mame4all is), so it won't be set. however if you have latest runcommand.sh it will ask you to choose a default on first launch if one is not set.

please can you confirm the second point by testing and I will add it as a choice.

ProxyCell, 2015-03-22:
i can confirm that SG1K games work with it, i added it's extension to the es_systems.cfg and it loaded up both congo bongo and lode runner

joolswills, 2015-03-23:
this is all complete. thanks.

closed 2015-03-23 by joolswills