RetroPie/RetroPie-Setup #755: updating emulation station gets stuck then crashes

InsecureSpike, 2015-03-23:
ok, i'm in the process of a clean install, via the current rpi2.img, but when i update emilationstation, from the set-up menus it gets stuck at 

[ 24%] Building CXX object es-core/CMakeFiles/es-core.dir/src/components/ImageComponent.cpp.o

and then nothing.... and the only way i can shut it down is to pull the power!!

any ideas?

joolswills, 2015-03-23:
your pi crashed ? out of ram ? probably bit of both.

not a retropie bug though - you might want to check free ram / decrease amount allocated to gpu.

but the binaries are up to date, so no need to build from source.

InsecureSpike, 2015-03-23:
ram set at 256

joolswills, 2015-03-23:
I would allocate less to the gpu if building from source - no idea what pi you have though, so little to go on.

joolswills, 2015-03-23:
i see you said rpi2 image, so I assume a rpi 2 ? so you have allocated 256mb to the gpu ?

should be enough to build emulation station, but it takes a while. overclocked ? decent psu / cable ?

The best fix is to install from binary - but the problem is related to your setup/rather than a retropie issue imho.

joolswills, 2015-03-23:
you can also try

```sudo MAKEFLAGS="-j1" ./retropie_setup.sh

InsecureSpike, 2015-03-23:
ok, i'll try some things out, and then get back,
thanks mate

InsecureSpike, 2015-03-23:
ok think i figured it out, i've been dong this..

http://www.htpcguides.com/power-2-5-hard-drive-with-raspberry-pi-b/

but this time i skipped the "Now update the Raspberry Pi’s firmware"
and just did the /boot/config.txt edit, and all seams good so far..

just doing a binaries update now

joolswills, 2015-03-23:
it would have been useful if you had mentioned you had a 2.5 disk plugged in. Glad you have it sorted.

closed 2015-03-23 by joolswills