RetroPie/RetroPie-Setup #758: Alphabetical es_systems.cfg

HerbFargus, 2015-03-24:
I think it would help make things a little more convenient for people if the es_systems.cfg file was in alphabetical order.

joolswills, 2015-03-24:
Completely agree. I think someone mentioned this before - maybe on the forum - I forget, but I also think it's useful. Needs me to spend a couple of hours with xslt to apply to the xml for sorting. Will get back to you :-)

joolswills, 2015-03-25:
I have added sorting by "name". had to decide whether to use name or fullname. neither is perfect, as for example "Amstrad CPC", name is cpc - should it be under A or C? If we used fullname, then things like "Sega Gamegear" will be under S rather than G. Perhaps some tweaks to the fullnames could fix this.

joolswills, 2015-03-25:
other options would be sorting by the platform field. Even sorting by name as currently, the list looks in a better order than it did before at least.

HerbFargus, 2015-03-26:
I personally prefer the fullname. If the name field matched the ROM folder exactly it might be more intuitive to some so long as the newer releases only have one rom folder per console rather than multiple rom folders per emulator (excepting maybe mame) but I suppose at this point it becomes personal preference. It probably just makes the most sense though to use fullname. In all reality a lot of the recent changes you have been making should eliminate the need for most people to even need to modify es_systems.cfg. The addition looks good

joolswills, 2015-03-26:
new releases have one rom folder per platform (apart from mame which is split still somewhat). It's just that if upgrading we don't remove any old rom folders, they must be removed manually.

the rom folders mostly match the platform names.

using the fullname seemed worse from my testing, due to some of the names having the company name first - initially i tried that then switched to name.

HerbFargus, 2015-03-26:
Sounds good, I trust your judgement on the matter- Can't please everyone anyhow :P Thanks for all the work!

closed 2015-03-26 by HerbFargus