RetroPie/RetroPie-Setup #763: unable to start emulationstation

Natherul, 2015-03-26:
I cant seem to be able to start emulationstation on my RPI 2 anymore, it seems that whatever I do I get that X is still running.... is there any way I can make it past this check either way as Im starting to dont care if x is running or not. I just want to start emulationstation!

joolswills, 2015-03-26:
please post the contents of the file /usr/bin/emulationstation

The error that X is running, means the desktop environment / Xorg is running. You need to exit from that first if it is, and disable it from starting up on boot perhaps ?

Natherul, 2015-03-26:
#!/bin/bash

es_bin="/opt/retropie/supplementary/emulationstation/emulationstation"

if [[ -n "2241" ]]; then
    echo "X is running. Please shut down X in order to mitigate problems with loosing keyboard input. For example, logout from LXDE."
    exit 1
fi

$es_bin "$@"

is whats in the file.
I dont want to disable x from starting as I use it as a server as well and I need my non tech savvy people to be able to turn the server off in case of an emergency (like thunderstorm, and they need a button to do that). 

Before sudo /etc/init.d/lightdm stop was enough to be able to get emulationstation to work but not anymore thus causing me this frustration... Also I aint that super good with linux yet but I am slowly learning.

joolswills, 2015-03-27:
There was a bug a short while ago that caused this problem. You must have updated for the few days the bug was there. please update retropie-setup script, and then do

```
sudo ./retropie_packages.sh emulationstation configure
```

tminus69, 2015-05-30:
Hello,

 I am getting an error saying line20: 2304 Illegal Instruction $es_bin"$@" I would like to try sudo ./retropie_packages.sh emulationstation configure rather than reinstalling another time. The system is telling me that sudo ./retropie_packages.sh command not found. Any help would be greatly appreciated.

closed 2015-05-30 by joolswills