RetroPie/RetroPie-Setup #764: [suggestion] License on readme and on main Github page

HerbFargus, 2015-03-26:
considering recent events it may be beneficial to the RetroPie Project to create a standalone license file in the source code as well as prominently displaying it in the readme just to eliminate any doubt that it isn't to be commercialised.

petrockblog, 2015-03-26:
That is a good idea. Given the current license information in the header of the files (see, e.g., https://github.com/petrockblog/RetroPie-Setup/blob/master/retropie_setup.sh), would you recommend a similar/comparable established license?

joolswills, 2015-03-26:
not sure if a creative commons licence would be suitable ? (I've used them for wiki's etc, but not code)

https://creativecommons.org/licenses/by-nc/3.0/deed.en_GB

the mame licence is a copyleft + non commercial one too.

http://mamedev.org/legal.html

joolswills, 2015-03-26:
Sorry - this creative commons one - http://creativecommons.org/licenses/by-nc-sa/4.0/

petrockblog, 2015-03-26:
Reading through the MAME license, I would have a good feeling to use the same license for RetroPie.

joolswills, 2015-03-26:
I'm happy with that - your current licence is pretty much equivalent.

Just for reading - http://en.wikipedia.org/wiki/MAME#License

The reasons for the non commercial clause for MAME is very much the reason why we need it too imho. I'm a big support of free software, however I also believe it's important in some cases to have a more restrictive licence to try and stop misuse, such as we are seeing.

HerbFargus, 2015-03-26:
I feel either of them will address the need accordingly but perhaps the creative commons license listed above is a more officially recognised license that has a more extensive legal backing to eliminate any discrepancy:

http://creativecommons.org/licenses/by-nc-sa/4.0/legalcode

We could have sort of a hybrid license that on the front end is very clear like the MAME license but legally is covered in the backend by the aformentioned creative commons license.

petrockblog, 2015-03-27:
At http://choosealicense.com the creative commons license is proposed for non-code projects. I have the impression that the creative commons licenses are not suitable for RetroPie.

HerbFargus, 2015-03-27:
Fair point. I guess the most important thing is that it is visible and easily accessible. After a perusal online, I think you're right. The MAME option seems to be most fitted to this project.

petrockblog, 2015-03-27:
I have created a proposal in a feature branch. 
 1. What do you think about the license? It is a (Q&A-) stripped down version of the MAME license?<br>https://github.com/petrockblog/RetroPie-Setup/blob/license/LICENSE.md
 2. What do zoo think about the header information that needs to be adapted later then?<br> https://github.com/petrockblog/RetroPie-Setup/blob/license/retropie_setup.sh

joolswills, 2015-03-27:
On 27/03/15 15:10, Florian wrote:
> I have created a proposal in a feature branch.
> 
>  1. What do you think about the license? It is a (Q&A-) stripped down
>     version of the MAME license?
>     https://github.com/petrockblog/RetroPie-Setup/blob/license/LICENSE.md
>  2. What do zoo think about the header information that needs to be
>     adapted later then?
>     https://github.com/petrockblog/RetroPie-Setup/blob/license/retropie_setup.sh

Do you think the clause regarding donations will work - wouldn't that
mean we breach the licence? - also make sure you/we retain copyright for
the sources so they can be relicensed as you need to (just in case).

apart from that looks good. If we include the licence file with the
project do we need to include it in the headers too or ? I suppose it
doesn't hurt to include it in the root scripts, but we wouldn't want it
in all of them I guess or it would get a bit bloat :)

Best Regards

Jools

joolswills, 2015-03-27:
I had missed the clause regarding central copyright. If you are happy with that then ok, but it might be easier to retain central copyright.

petrockblog, 2015-03-27:
Good points! I have modified the LICENSE.md and header information accordingly now: https://github.com/petrockblog/RetroPie-Setup/blob/license/LICENSE.md

I still think that we should have a brief hint to the license in each file. What do you think of the updated header information in https://github.com/petrockblog/RetroPie-Setup/blob/license/retropie_setup.sh#L3?

joolswills, 2015-03-27:
Header looks good. So if we include that on the two main scripts, and the code in scriptmodules, but not every script module would that suffice ?

In the Q/A the bit that says - Q. How do I obtain a license to the RetroPie source code? A. You can't. See the Copyright section above. can probably be removed now ?

HerbFargus, 2015-03-27:
I think it looks really good :) copyright headers are good for the main scripts and could be foregone in some scriptmosules as long as the main license specifies that all scripts are included under this license. Just needs to be clear they are all covered. Good work :)

petrockblog, 2015-03-27:
I have updated all files and already merged it into default (fc5a1ac110b4e2770deb1bae3520f52b01d9f1c9). I think this issue can be closed now or is there anything left to be discussed about this topic?

joolswills, 2015-03-27:
look good to me. Cheers!

HerbFargus, 2015-03-27:
Looks good to me. Great work!

closed 2015-03-27 by HerbFargus