RetroPie/RetroPie-Setup #765: RetroPie 3.0 Beta Romservice

HerbFargus, 2015-03-26:
update says:
* Updates to the usbromservice. If you want to sync rom folders it now
requires a folder in the root of the usb stick called “retropie”. The
roms will be synced from a sub folder called roms. It also can
backup/restore your custom emulationstation gameslists / data.

but upon testing it worked as normal without the need for the "retropie" folder. I tested it with "retropie", "RetroPie", and each time I did it, it didn't matter because it created a roms folder in the base of the usb directory rather than the retropie directory.

HerbFargus, 2015-03-26:
Update: 

upon updating the setup script and re enabling the romservice the "retropie" folder is indeed a necessity and works as described in the changelogs. I guess the functionality was updated after the download 3.0 beta download was published.

closed 2015-03-26 by HerbFargus

joolswills, 2015-03-27:
It was done before, but I forgot to apply the update before I put the image together. I've corrected that so it should be good for the next release.

joolswills, 2015-03-27:
thanks for the heads up.