RetroPie/RetroPie-Setup #766: Game specific config files for retroarch

janezkranjc, 2015-03-27:
Hello!

Would it be possible to implement a separate config file for certain roms?

http://blog.petrockblock.com/forums/topic/game-specific-config/ <- this guy proposed a pretty good solution but it doesn't work because he can't put %ROM% twice in the command.

joolswills, 2015-03-27:
Can you give me an example use case ? it could be done via runcommand, but what would you need to change on a per rom basis ?

janezkranjc, 2015-03-27:
I am using lr-genesis-plus-gx which allows me to play 4 player games such as general chaos and fifa with 4 controllers if I add this setting to the megadrive retroarch.cfg:

input_libretro_device_p2="1025"

However, if I have this setting enabled all other games that do not support multitap do not work as controllers are no longer detected by the games. Whatever I press even on controller 1 doesn't work.

So I would need to have a separate config for multi-tap games (general chaos, fifa, nba jam) and regular one or two-player games

joolswills, 2015-03-27:
ill see what I can do.

janezkranjc, 2015-03-28:
Thanks for merging!

closed 2015-03-28 by janezkranjc

joolswills, 2015-03-28:
no probs. I've made a couple of changes - to use the dialog editbox rather than nano, and I removed the additional regular expression, as it can just be appended to the conf variable..

janezkranjc, 2015-03-28:
It looks really cool now! And it deletes the config file if it's empty, very clever :)