RetroPie/RetroPie-Setup #767: mupen64plus build broken

joolswills, 2015-03-27:
@gizmo98 

gles2n64 isn't building. also the rice plugin (at least for rpi1)

I'm working on some fixes using sed in the build script-  I commented on a change for the gles2n64, once I have it built, I can feed back some fixes upstream, unless you have time to do it (you are probably more familiar with this than me)

joolswills, 2015-03-27:
moving the -flto in the gles2n64 makefile to OPTFLAGS fixes that. Recent include changes in the upstream rice plugin have broken building it on the pi. I've forked it and will revert those changes just to get a working build, but it will need to be fixed upstream properly of course.

joolswills, 2015-03-27:
https://github.com/mupen64plus/mupen64plus-video-rice/pull/52#issuecomment-87059803

joolswills, 2015-03-27:
ok all fixed. thanks for the help.

closed 2015-03-27 by joolswills

gizmo98, 2015-03-28:
I ported mupen64plus-pandora's gles2rice plugin today. It has neon assembler and shading looks as good as gles2n64. Mario 64 looks much better. Will try to polish and setup before 3.0 final will be released.