RetroPie/RetroPie-Setup #772: mupen64-plus questions.

joolswills, 2015-03-28:
Hi, @gizmo98 

So just testing mupen64plus - the gles2n64 seemed to render at native video resolution, whereas the rice plugin was fixed to 640x480. Is it possible to make them consistent so they both for example render at the same resolution?

I'm guessing there is no way currently to have them use hardware scaling to avoid having to switch mode to get the rice plugin to go fullscreen ?

I was thinking about the audio too - it defaults to hdmi, wonder if there is a way to make this automatic - not looked at the code yet.

gizmo98, 2015-03-28:
You can test "ricrpi/mupen64plus-video-gles2rice pandora-backport": https://github.com/ricrpi/mupen64plus-video-gles2rice/commit/fd7db65ae2530ab923fa10507c6e925ac962fb84

This branch is only some hours old. Native resolution rendering, keep aspect ratio and better shading is already working. RPI1 is not supported atm.

To get hardware upscaling we need a modified sdl2 package. src and dst rect are equal atm:
https://github.com/spurious/SDL-mirror/blob/master/src/video/raspberry/SDL_rpivideo.c#L249

joolswills, 2015-03-28:
yeh - I should look at that and add some options. we could use some simple env variables to specify/force render res

joolswills, 2015-03-28:
nice work btw :)

gizmo98, 2015-03-28:
Ok. platform RPI1 and RPI2 should work now.

gizmo98, 2015-03-29:
The new pandora plugin has a issue with Zelda majors mask. Screen hangs after some minutes. I increased GPU mem to 386MB, disabled cpu governor switching and updated firmware and the problem is gone. Don't know if it is a memory or temperature problem atm.

gizmo98, 2015-03-29:
@joolswills. If pandora backport runs good we should use it as default video plugin. gles2n64 runs a bit faster but game compatibility is bad. Pandora-backport has neon optimizations as well and in my opinion visual quality is on par now.

joolswills, 2015-03-29:
sounds good.

McPace, 2015-04-02:
Hi @gizmo98, I was wondering if you could help a noob out who has time to play games and test things. How do I install this pandora video plugin to help test?
"You can test "ricrpi/mupen64plus-video-gles2rice pandora-backport": ricrpi/mupen64plus-video-gles2rice@fd7db65"

gizmo98, 2015-04-03:
Start RetroPie-Setup. There is a menu item in RetroPie SD Card image 3.0 beta.  Select experimental packages and select mupen64plus-testing. Afte installation you have more n64 emulator options if you press "x or m" after game selection.

McPace, 2015-04-03:
Thanks @gizmo98  I had previously installed the experimental, so I updated them (I saw lots of warnings scroll past during install) and after install I didn't see an option labeled with "pandora" specifically so I played with the gles2rice testing option. It failed (video plugin not found, closed rom) on most ROMS. Mario Kart played well. 

I assume if I rerun the experimental install script I'll get the most up to date to see if anything has changed?

closed 2015-10-10 by joolswills