RetroPie/RetroPie-Setup #774: .7z files and libretro cores

Tele42, 2015-03-29:
All libretro cores that accept .zip files potentially also accept .7z archives via the same mechanism. Some testing is required before adding this to the relevant parts of platforms.cfg.

free5ty1e, 2015-04-03:
This would be quite awesome.

joolswills, 2015-04-04:
I believe the 7z support was somewhat more limited than the zip in regards to supporting only single files in an archive etc. this may have changed, but please do try it and let me know. Thanks.

free5ty1e, 2015-04-04:
Wait, there's no need to have games all individually zipped up?  Retroarch does something sensible when it is passed a ZIP archive containing multiple ROMs?  :o

joolswills, 2015-04-04:
I believe there were initially more limitations to the 7zip support than zip such as directories not being supported. this was some time ago however, and it may have been resolved since - hence it needs some testing.

joolswills, 2015-04-04:
I'm not sure how *smart* retroarch is when it comes to zips either :)

HerbFargus, 2015-04-04:
I've usually had more trouble with zipped folders than not, I've always just unzipped my files first and never had any problems.

arthurlutz, 2015-08-04:
+1 on this

joolswills, 2015-10-10:
Has anyone had time to test this ? It needs some user testing before I change anything.

HerbFargus, 2015-10-10:
tested with nes and snes emulators, and a no go- its likely the same for the rest of the systems as well. If people can prove support for .7z on the pi with retroarch we can look on adding them but for now it looks like it isnt viable.

closed 2015-11-05 by joolswills

davidjb, 2016-02-16:
Tested with Megadrive / Genesis and Retropie v3.5 and nothing is listed within Emulationstation if inside a .7z file.  Zip files partially work, with the only the final file (by filename, lexographically it seems) unzipped and loaded; all other files present in the `.zip` are ignored.  This unzipped file remains in the folder after exiting.  

+1 for convenience, though, if both .zip and .7z could be supported; but it's not that difficult to just unzip first.

dankcushions, 2017-08-31:
I think this needs revisiting now that we have added .zip support across all (?) libretro cores. I think they are functionally identical now?

care must be taken on the arcade cores which have their own loaders. lr-fbalpha supposedly supports 7zip, but the others maybe not.

joolswills, 2017-08-31:
@dankcushions what needs revisiting ? .7z was added to most of the platforms that have a libretro emulator.

dankcushions, 2017-08-31:
Ah, good! Then only the help and wiki needs updating. Eg:
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-snes9x2010.sh#L14 
https://github.com/retropie/retropie-setup/wiki/Super-Nintendo-Entertainment-System

joolswills, 2017-08-31:
Yeah. Wanted to get any bug reports etc first but can probably be added now.

joolswills, 2017-08-31:
There may be some platforms missed also (and some that as you say natively load that don't work). Because we have per system extensions for ES, there could be some mistakes also.