RetroPie/RetroPie-Setup #776: Wired PS3 Controller Setup Only works in Shell not In EmulationStation

alph486, 2015-03-29:
Hi there,

I'm using the new 3.x Beta image of RetroPie. I Followed this video: https://www.youtube.com/watch?v=DG0_Ig1yLUE to setup my PS3 controller. It worked fine when I'm operating in the Shell environment, the driver recognizes the device etc. However, as soon as I launch into emulationstation to go and add my input device, the software will not detect that it has been connected. If I try to reconnect the device, it will flash with all of the player "slot" lights and never turn on the solid "1" like you would expect. If I exit back into the linux shell and reconnect it will work fine.

Any thoughts on a configuration issue or is this a bug in the new one?

joolswills, 2015-07-13:
Please can you test with the latest beta - it is hopefully resolved now.

joolswills, 2015-08-10:
closing due to no feedback.

closed 2015-08-10 by joolswills