RetroPie/RetroPie-Setup #777: pifba: PS3 dpad not working

m42i, 2015-03-30:
I couldn't use the dpad on my PS3 controller in pifba even when I've added them to the .cfg. 

It works using this repo:
https://github.com/digitalLumberjack/pifba

This seems to be the corresponding commit:
https://github.com/digitalLumberjack/pifba/commit/97fd778de4d7accee786ed5089823798017573e3

HerbFargus, 2015-03-31:
The repo above also supposedly enables 4 players instead of being limited to 2 players like the current build- Might be worth looking into.

joolswills, 2015-07-17:
We should have 4 player support in our pifba now. Not tested though.

HerbFargus, 2015-07-17:
I updated the setup script and installed from binary- it acts like its loading but then it exits right back out. runcommand gives this error- this game was previously working with pifba before the update (cadillacs and dinosaurs)
```
/opt/retropie/supplementary/runcommand/runcommand.sh: line 684:  2847 Segmentation fault      /opt/retropie/emulators/pifba/fba2x "/home/pi/RetroPie/roms/fba/dinoj.zip"
```

joolswills, 2015-07-17:
https://github.com/RetroPie/RetroPie-Setup/issues/908

joolswills, 2015-07-17:
(otherwise it gets confusing with multiple issues under different tickets etc)

HerbFargus, 2016-03-15:
would someone with a ps3 controller be willing to test the latest pifba binaries to see if this is still an issue?

joolswills, 2016-08-23:
I have tested and checked the code - it is hardcoded to only support axis for movement. the recalbox fork has additional inputs - I might look into borrowing parts of that.