RetroPie/RetroPie-Setup #778: WIreless Xbox 360 controls not responding in shell

McPace, 2015-03-30:
Using the 3.0 BETA image the new retropie menu is cool and useful as well as the video / emulation pre rom loading options. Unfortunately when I enter any of the script options the xbox 360 wireless controller doesn't work on the menus, so you have to plugin a keyboard again. This makes the new menu option require two inputs, controller when you get bounced into Emulation Station and a keyboard when you're in setup / video option menus. Can we enable the 360 inputs for use in the setup/config menus?

joolswills, 2015-03-31:
We might be able to do a generic solution using joy2key, but there will still be stuff that won't work via keystick, such as configuration items that require text to be entered. Easiest way is to get a cheap wireless/bluetooth mini keyboard and have that connected for when you need to do configuration.

McPace, 2015-03-31:
A generic solution for up, down and enter would be awesome! And thank you very much for all your work on this.

joolswills, 2015-04-04:
One of my main problems is I don't have a usb controller to test. Not even one!

So this might have to wait until I get one.

McPace, 2015-04-04:
That is sad. Send me your address, I'll mail you one.

joolswills, 2015-04-04:
Thanks - please email me @ buzz [at] exotica.org.uk and I will send you my address.

I actually purchased a snes style usb controller earlier today - so I will have something to use once it arrives (from japan). There are a few things I would like to test with a 360 controller, so if you had one of those I would gladly accept - alternatively I can probably borrow one from someone locally. Cheers.

joolswills, 2015-06-19:
I have added initial support for controlling retropie-setup with a joypad.

if you update your script you will find a new item "joy2key" under the supplementary menu. Install that, and then next time you launch retropie-setup you may be able to control the menus with your joypad. It has some variables though so we might need to do some adjustments for your setup. Try it first, and see if you can move up/down/left/right. button 1 (a) maps to enter, and button 2 (b) maps to space.

joolswills, 2015-06-19:
@HerbFargus - If you have time to test this out also that would be much appreciated - thanks. :)

HerbFargus, 2015-06-19:
Will do

HerbFargus, 2015-06-19:
Works great! This is just a nitpicky thing and not a super big deal, but it may be more intuitive to people if the mappings for enter and space corresponded to their A and B keys mapped in inputstation (B would be enter, A would be space)- In my case my controller mapped enter to button 0 (which is X) and space to button 1 (which is A).

joolswills, 2015-06-19:
I can't really map it like that currently, but perhaps in the future. It's not really supposed to replace keyboard usage either, but rather a helper (some config stuff requires keyboard for entering things).

It will work for entering the menu before launching with my next commit.

joolswills, 2015-06-19:
runcommand now has basic joypad support too. if you update the retropie-setup manually you will need to reinstall runcommand.

HerbFargus, 2015-06-19:
Well that was quick. Works great :)

HerbFargus, 2015-06-19:
works great as long as you don't use the keyboard... It seems to override the keyboard which I guess is fine as long as only controllers are being used.

joolswills, 2015-06-19:
oh it shouldn't do that - I may have introduced a minor bug. will check.

joolswills, 2015-06-19:
on my system both keyboard and controller is working fine. The code shouldn't touch the keyboard at all, it just sends keypresses to the terminal when joystick buttons are pressed. Are you able to help me reproduce? are you talking about runcommand or in retropie-setup or both ?

HerbFargus, 2015-06-19:
So this is only in runcommand, the terminal works fine

joolswills, 2015-06-19:
I can't reproduce it - it works fine here :/

HerbFargus, 2015-06-19:
just for reference I updated the setup script, then I ran sudo /opt/retropie/supplementary/runcommand/runcommand.sh to install it. was I supposed to install it differently?

joolswills, 2015-06-19:
/opt/retropie/supplementary/runcommand/runcommand.sh is the old location of the script in the repository. No longer lives there. to install do

```
sudo ./retropie_packages.sh runcommand
```

joolswills, 2015-06-19:
sorry misread. /opt/retropie/supplementary/runcommand/runcommand.sh is the installed location of the script. which will be an older version if you didn't update.

HerbFargus, 2015-06-19:
ok yeah the sudo ./retropie_packages.sh runcommand solved it :) we're good.

joolswills, 2015-06-19:
:) I've made another update to fix a harmless error message if no joystick was connected.

McPace, 2015-06-22:
This is so cool! Thanks for adding this. On my Wireless Xbox 360 controller, the directional is setup on the analog stick instead of the d pad (which would be fine) but the sensitivity is so high that the option highlighter flies around as you move the stick. It's giving continuous input. It makes it too hard to land on the option you wanted to. Maybe using the dpad would give it more of the expected one button press down equals the option moving down just one.

joolswills, 2015-06-22:
yeh, it's preliminary support and it doesn't have any configuration yet, so it will only work where the first axis is digital such as a snes type controller. Works well on one controller of mine, but not on the XBox (yet).

McPace, 2015-06-22:
No worries, I'm sure you super smart people will smooth it out. I'm glad the option is there. Much easier than grabbing a keyboard all the time.

joolswills, 2015-06-23:
I have rewritten a simple joystick to keyboard press program in python, which should work better with analog controls. If you update retropie-setup (from the gui), it should install the new code automatically (you will need to restart retropie-setup)

McPace, 2015-06-23:
Is it called xarcade2jstick service now?

McPace, 2015-06-23:
Activating xarcade2jstick didn't work for me. No response at all from wireless Xbox 360 controller

joolswills, 2015-06-23:
no. it's called joy2key - did you update retropie-setup script from the retropie-setup gui, and then restart ?

joolswills, 2015-06-23:
actually. it will require doing that twice as the script needs to be restarted and then updated again.- or manually re-running runcommand from the supplementary menu after updating, and restarting.

McPace, 2015-06-23:
Ah ok, I'll try again later. I updated retropie setup script once and restarted. joy2key was gone from the supplemental menu but xarcade2jstick was there and I hadn't remembered seeing that so I thought it was renamed. I'll update again and see if joy2key reappears.

joolswills, 2015-06-24:
There is no menu item for it now - it's no longer something that needs compiling/installing so it's done automatically as part of the runcommand functionality

joolswills, 2015-07-01:
did you get around to testing this ?

McPace, 2015-07-01:
I tried again last night but I think I need to upgrade again.

joolswills, 2015-07-01:
I can roll you an updated image with it working if that would help.

McPace, 2015-07-01:
That would. Each time I update from binaries or such I don't seem to get updates. Is there a version number hidden somewhere to know if you're running the latest? Maybe there could be?

joolswills, 2015-07-01:
It's somewhat complicated to have that - we would need to have some mechanism of "locking" emulators to a specific version etc. right now many of the retroarch emulators will build from binary - so you have two things - the versions of the retropie-setup script, and the versions of the emulators etc. Basically, the image releases are where we are currently with the script, usually with the emulators updated to the latest version (such as retroarch ones). Other emulators are locked already at specific versions - such as hatari, or atari800, and uae4all.

joolswills, 2015-07-01:
if you update from binaries you will get the latest binaries for the emulators - which get built every so often (a couple of days ago was the last time). You should update retropie-setup script first though before doing that - which makes sure you have the latest versions of the configuration scripts.

McPace, 2015-07-01:
Right so just hitting update retropie setup script should take me to the latest version right that houses this feature. I guess I'm unclear of when I update the setup script and should see new features vs updating all emulators. Yes I always do upgrade the setup script first... but it never takes very long and it isn't clear that things are new. I'll go again today after work.

joolswills, 2015-07-01:
the feature is built into the script code rather than a separate binary - updating it from the GUI also runs the runcommand module configuration which I attached this functionality to (as it's connected to input during the launch menu too etc). To get new emulators you still need to update the binaries though.

McPace, 2015-07-02:
I ran the retropie setup script update
rebooted
went into the retropie setup menu again, no response from controller at all.
ran update from binaries
rebooted
went back to the retropie setup script, no response from xbox 360 wireless controller

Interesting note, after update from binaries, Emulation Station asked to reconfigure controller and right trigger wouldn't register.

joolswills, 2015-07-02:
if you drop to a terminal what is the output of

```
ls -l /dev/input/js*
```

I have similar problems with shoulder buttons on my original xbox controller. I believe running xboxdrv may help. You know the quickest fix here is to purchase a £10 usb controller that works out of the box :) xbox 360/ps4 controllers can be tricky.

joolswills, 2015-07-02:
Just to confirm - on my original xbox controller I need to install the "xbox 360 userspace driver" for it to work reliably - the option is on the supplementary menu in case you have not installed it.

McPace, 2015-07-02:
Went to supplemental menu and installed xbox 360 drivers.
Reboot
Ran Emulation Station controller config again, trigger was recognized!

Dropping to terminal and running ls -l /dev/input/js* gives me:
crw-rw---T 1 root input 13, 0 Jan 1 1970 /dev/input/js0
crw-rw---T 1 root input 13, 5 Jul 2 01:48 /dev/input/js5
crw-rw---T 1 root input 13, 6 Jul 2 01:48 /dev/input/js6

joolswills, 2015-07-02:
and does the joypad work in retropie-setup yet ?

if not please run the script update option in retropie-setup again - then restart. Working ? If still not please drop to a terminal and do

```
/opt/retropie/supplementary/runcommand/joy2key.py /dev/input/js0 4c0a 520a 550a 440a 31 32 33 34
```

and see if U D L R and 1 2 is outputted for dpad / buttons

joolswills, 2015-07-02:
(I have confirmation from another user it is working with their 360 controller, although they have issues with retroarch currently)

McPace, 2015-07-02:
No response still in RetroPie Setup menu.

Wondering if it wasn't running b/c I had the keyboard plugged in as well, I unplugged the keyboard while on the setup screen and this error popped up:
Traceback (most recent call last): File "/opt/retropie/supplementary/runcommand/joy2key.py", line 49, in <module> event = js_fd.read(event_size) IOError: [Errno 19] No such device

McPace, 2015-07-02:
If it is working for others, I can wait to install beta 5/ 3.0 from a fresh image

joolswills, 2015-07-02:
not sure why unplugging the keyboard would affect the joystick inputs. maybe it reset them too or you jogged it whilst unplugging - for whatever reason /dev/input/js0 diappeared when you did it (joystick device) - please post the results of the other test I posted above.

joolswills, 2015-07-02:
(and try without keyboard too in case it is somehow related)

joolswills, 2015-07-02:
the above error btw meant the joy2key code was running and was reading data from a joystick device (until it was interrupted). Did you have multiple joysticks connected or just one ?

McPace, 2015-07-02:
Tried without keyboard, still no response.

Above test (/opt/retropie/supplementary/runcommand/joy2key.py /dev/input/js0 4c0a 520a 550a 440a 31 32 33 34) just shows a blinking terminal input. I imagine it's supposed to register the keystrokes?

joolswills, 2015-07-02:
if you press something yeh. very odd. and if you use

McPace, 2015-07-02:
Yes keyboard hits are logged

joolswills, 2015-07-02:
if you press something on the joystick it should output some letters. very odd. and if you change  /dev/input/js0 to  /dev/input/js5 (or js6 as listed above ?)

McPace, 2015-07-02:
No response on 5 or 6, keyboard registers on both

McPace, 2015-07-02:
Sorry :(

joolswills, 2015-07-02:
no idea then. hmm. does "jstest /dev/input/js0" work ?

McPace, 2015-07-02:
Wow. I've found the matrix. :) Yeah that shot out a ton of stuff

joolswills, 2015-07-02:
then it's very odd why my joystick reading code isn't working - tested on two controllers here, and another 360 joypad user has it working. Here's a new image should you want to test from a clean install - http://malus.exotica.org.uk/~buzz/pi/retropie-v3.0test-rpi1.img.gz (for rpi1, but should work on rpi2)

[4 more replies]

closed 2015-07-13 by joolswills