RetroPie/RetroPie-Setup #781: USB drives are not being mounted with exec privileges.

redhalo, 2015-04-01:
USB drives are not being mounted with exec privileges. In my use case, I am storing all roms on an external hdd and symlinking it in place of the default rom folder. The issue is that I cannot run bash scripts out of these folders, Scumm and ports scripts for example. The work around I have in place is to issue a "sudo mount -o remount rw,exec,users,auto /media/usb". Could we have drives mounted with exec privileges by default?

joolswills, 2015-04-02:
USB drives are not mounted by default unless using the usbromservice/copyroms functionality, where it is mounted synced and unmounted. How are you mounting your drive ?

redhalo, 2015-04-02:
Plugging it in. I've deleted 01_retropie_copyroms from /etc/usbmount/mount.d/ but it still mounts after reboots. So...I don't know, it just does. That and the symlink to the hard drive are the only manual changes I've made to a fresh 3.0 beta image install.

joolswills, 2015-04-04:
what are the contents of your /etc/fstab ?

joolswills, 2015-04-04:
You are likely right btw, as I think I've had a brain glitch - the usbmount package should indeed by default mount drives automatically. it's just that our sync service will unmount after it completes, so if that is removed, then it will remain mounted.

to set default mount options for the usbmount service, just edit /etc/usbmount/usbmount.conf and remove the noexec parameter, or remove the usbmount package and mount manually from fstab

joolswills, 2015-04-04:
you can also disable the service via that config file without removing it.

redhalo, 2015-04-04:
Cool, that seems to do the job. Thanks.

joolswills, 2015-04-04:
welcome. apologies for the confusion.

closed 2015-04-04 by joolswills