RetroPie/RetroPie-Setup #782: Hatari performance suggestions and SDL2 builds

amigagamer, 2015-04-01:
Few suggestions for improving hatari performance.

Since we can use buzz SDL_dispmanx to handle the scaling , launch hatari with   hatari --zoom 1 to disable the software "double sizing" the image.

loading with "--timer-d 1"  Gives a good performance increase with a slight drop in compatibility.  This can be toggled in the menu once hataris loaded, but maybe a good performance option to set as default (esp for RPI1 users)

Im using both  with  "hatari --timer-d 1 --zoom 1"

on a further note its worth looking at the mercurial repo.  the latest dev builds support SDL2 scaling and vsync.  http://hg.tuxfamily.org/mercurialroot/hatari/hatari  
it also seems to have resolved the annoying bug with menu buttons "sticking" on the file browser
needs configuring with ./configure --enable-sdl2

All tested on Raspberry pi 2
performance references http://hg.tuxfamily.org/mercurialroot/hatari/hatari/raw-file/tip/doc/manual.html#Performance

joolswills, 2015-04-02:
thanks.

even without dispmanx sdl, a 320x240 framebuffer mode would also use hardware scaling, so I will add that as a default. there can be a performance penalty running with the dispmanx sdl backend as you probably know due to the vertical blank wait.

I'll look at building with sdl2.

joolswills, 2015-04-04:
you can also try with "--compatible 0" which makes a significant speed-up on my pi1.

joolswills, 2015-04-04:
not had a chance to look into sdl2 currently, but the above changes should help.

amigagamer, 2015-04-04:
Good idea on the compatible vs fast profiles.   Since they all use the same hatari.cfg (which can be saved within the emulator) it might be an idea to add --compatible 1 and --timer-d 0 to the "compatible" and "compatible-borders" profiles.  Otherwise people switching between profiles may be stuck in "fast" mode due to saved options in hatari.cfg

joolswills, 2015-04-04:
good point - ill do that.

amigagamer, 2015-09-15:
Hatari 1.9.0 final now released with SDL2 support included  http://hg.tuxfamily.org/mercurialroot/hatari/hatari/log/978142ac82ff
 
source code at http://download.tuxfamily.org/hatari/1.9.0/hatari-1.9.0.tar.bz2   

build with ./configure --enable-sdl2

joolswills, 2015-09-15:
thanks - yeh saw that the other day - will be updating it when I have a chance.

joolswills, 2015-10-10:
This is now done. Binaries will be updated shortly.

closed 2015-10-10 by joolswills