RetroPie/RetroPie-Setup #783: RDP/VNC not working if Emulation Station is running

diogomsantos, 2015-04-02:
Like the title says, when ES process is running it is not possible to connect to the Pi via RDP or VNC.
I think it is because this line in the Emulation Station auto start:
echo "[ -n \"\${SSH_CONNECTION}\" ] || emulationstation" >> /etc/profile

If i change it to echo [ -n \"\${SSH_CONNECTION}\" ] || emulationstation &, it works.
I didnt find any drawback in doing this, but maybe there is ?

If not can you please change it? Or change the autostart script to any other way you think its best to not prevent a RDP/VNC connection if ES is running.

Thanks

joolswills, 2015-04-02:
you are running / booting to a desktop environment ? if so, you probably don't want emulationstation autostarting at boot anyway, so you should disable that from the script.

If not, please provide more detailed information regarding the setup - including exactly which protocol/server software so I can reproduce it, and come up with a workaround (Not sure that backgrounding emulationstation is ideal in regards to other functionality )

diogomsantos, 2015-04-02:
No. I am booting the system, latest Raspbian and updated, with a recent instalation of the Retropie suite (2015-03-27), directly to Emulation Station. Thats exactly what i want. But i have LXDE that i use frequently and sometimes i use RDP to access it remotely.
This is in a RPi 2 and I am not using rpi-update or any other core component changed.
I always had this problem, since ES 1, and using the fix I said but only now i decided to report it. 

Its easy to reproduce it. Just install xrdp package from the Raspbian repo and make sure the service is running. Have ES open also. Then, connect to it from another machine (Remmina or Windows Remote Desktop for example). Login and the screen will stay all grey and not working. Then, if i kill emulationstation, the connection starts working. If I do this with ES closed there is no problem. The same happens if you install tightvncserver and have it started.

I can paste some logs if you want, not sure which ones could be usefull in this case.

joolswills, 2015-04-02:
our emulationstation launch script checks if the desktop is running, and shouldn't even launch if it is - they don't play nicely together afaik hence this was done. I thought with your original post you were saying that it works if emulationstation is running but backgrounded from /etc/profile - is that still the case or are you saying RDP doesn't work when ES is running fullstop ?

I didn't think /etc/profile would be run for a remote gui session, but perhaps it is, and a check could be made. I'll debug a bit.

joolswills, 2015-04-02:
ah is xrdp standalone ? Might be that we need to check for that as well to avoid trying to launch es.

joolswills, 2015-04-02:
What happens if you change the line in /etc/profile to

```
[ "`tty`" = "/dev/tty1" ] && emulationstation
```

diogomsantos, 2015-04-02:
Works the same way as 
[ -n \"\${SSH_CONNECTION}\" ] || emulationstation 
and the remote conections works to. Nice!

I will test a bit more to check if theres is any problem with this change.

joolswills, 2015-04-02:
thanks - appreciated.

diogomsantos, 2015-04-03:
I can confirm. There is no problems with this change. Thanks again

closed 2015-04-04 by joolswills