RetroPie/RetroPie-Setup #784: Some ADVMAME games can't be stretched to the right vertical resolution

andrewsil1, 2015-04-02:
There were some changes recently I think to how Advmame is getting built, with some patches to do all the hw scaling in the Pi's frame buffer (Pi2 in my case).  Unfortunately there are some games (Blasteroids is a good example) where the internal resolution of the game is something funny like 640x240, and I imagine it originally relied on the hardware to double the vertical resolution so as to come out at the intended aspect ratio.

The advmame UI options for "Resize" used to be able to address this, but with these recent patches, there appears to be no way at all to get games of this sort to display at the correct aspect ratio now, it's terminally squashed.

Lemme know if you need any additional info.

Andy

joolswills, 2015-04-03:
I switched back from SDL to using the framebuffer directly as it affected performance (noticeable on a pi1). I adjusted the framebuffer code, so rather than trying to generate a suitable video mode and calculating values from pixel clock etc (which caused problems creating correct modes for some games), it just accepts the game resolution - but this assumes the games use square pixels. The original mode generation probably added some different modes with different aspect ratios to sort out games like this - and the sdl code., which isn't the case now. I'll see what I can do. 

In the meantime you can switch to sdl perhaps by changing device_video and device_keyboard to sdl

Which other games are affected that you noticed ?

andrewsil1, 2015-04-03:
I'll have to see what other ones I can find.  I went through a little over a hundred last night and actually Blasteroids was the only one I found that was noticeably mangled to the point of unplayability.  I'll do some more searching for others- perhaps there are other Atari games of the same era built on the same basic design.

joolswills, 2015-04-03:
"noticeably mangled to the point of unplayability" = you mean it runs at half vertical height right ? Unless you are referring to another problem I can't see (I prefer technical details over dramatic descriptions ;-) )

joolswills, 2015-04-03:
I think if I add two additional modes that double width and double height, it should sort it.

andrewsil1, 2015-04-03:
Sorry, I live for dramatic descriptions. :) Yes, it's compressed to half the intended vertical aspect ratio.

joolswills, 2015-04-03:
:) 

I'm digging now. I can easily add some manual modes which would suffice, but I'm just seeing if I can work out the resolution automatically etc.

andrewsil1, 2015-04-03:
Found a pretty brilliant DB of Mame roms sorted by resolution. :)

http://mamelist.bytenoise.co.uk/display/

joolswills, 2015-04-03:
thanks.

added two modes that have doubled x and doubled y resolutions so you can manually switch the video res to fix blasteroids (and other similar games) - binaries will be updated shortly.

andrewsil1, 2015-04-03:
Is there a quick route to redownloading the binaries without nuking things I have customized in the config? (e.g. the advmame.rc files, plus the emulators.cfg which I have calling a joystick configuration script on a per-rom basis prior to launching the emulator).

joolswills, 2015-04-03:
the configs shouldn't be nuked - but our specific settings may get reset to defaults such as samplerate and other bits we set - just install from binary in menu 5. Backup your configs just in case though.

andrewsil1, 2015-04-03:
OK.  I’m guessing the binaries aren’t quite done yet because after redownload I’m not getting the generate-double-y or double-x modes.  I’ll try again a bit later.

joolswills, 2015-04-03:
pi2 binaries are updated now

andrewsil1, 2015-04-03:
Fixed! (Have to choose double-y mode and also Integer stretch in ADVMAME menu.)

closed 2015-04-03 by andrewsil1

joolswills, 2015-04-03:
the default scaling is "mixed" which it works with too - or it seemed to for me.

andrewsil1, 2015-04-03:
Yes, that works too.