RetroPie/RetroPie-Setup #790: Wrong file path in the quick and dirty method

joydisee, 2015-04-04:
hey !

So I've been trying to configure a wiimote for my Retropie install using the *Quick and dirty method* (given here: https://github.com/petrockblog/RetroPie-Setup/wiki/Wiimotes-with-classic-controllers#quick-and-dirty-wiimote-configuration), and I kept having the program say, right after it connected to the wiimote that the file */home/pi/bin/wiimote.input* could not be found. Well that seemed kind of obvious since I have never created such a file,

I then realized that the script */home/pi/bin/attachwii.sh* was looking for this file: */home/pi/bin/wiimote.input*, however, when the user is asked in the tutorial here (https://github.com/petrockblog/RetroPie-Setup/wiki/Wiimotes-with-classic-controllers#correct-usage-of-wminput) to create the mapping script, he is asked to register it here: /home/pi/mywminput with a different name than the one *attachwii.sh*. So I eventually changed the name and path in attachwii.sh, which eventually make it work now.

Did I do something wrong or is there a problem with files' paths and names ?

Thank you for your time and great job !
ilyass

(@usag1e this might help you too :) )

joolswills, 2015-04-09:
the wiki is likely incorrect - the content is mostly user generated. it also gets out of date quickly.

joolswills, 2015-04-09:
feel free to correct it if you have worked it out.

closed 2015-06-20 by joolswills

idflood, 2015-12-08:
I've edited the wiki since this fixed my issue too, thanks @joydisee

HerbFargus, 2015-12-08:
Glad to see at least one person is being proactive. Thanks for making the path better for others that follow.

joydisee, 2015-12-08:
Sorry for never doing that @joolswills , I forgot about it since I guess I spent my time working on the install and configuring stuff right after.
Happy it helped you too @idflood, and thank you for being proactive :+1: 
Thanks for your comments about pro-activity @HerbFargus it was quite relevant and helpful.

Cheers,
thanks to the team for the great work.